SICKAG/sick_scan_xd #262: sick_scan_xd generic version keeps failing to build on docker.

AlecLozano, 2024-01-24:
Im following the instructions from [here](https://github.com/SICKAG/sick_scan_xd/blob/master/test/docker/docker.md) but docker keeps throwing out this error:

2024-01-23 23:43:00 ../../build_linux/CMakeFiles:
2024-01-23 23:43:00 total 128
2024-01-23 23:43:00 drwxr-xr-x 11 root root  4096 Jan 24 04:31 .
2024-01-23 23:43:00 drwxr-xr-x  1 root root  4096 Jan 24 04:31 ..
2024-01-23 23:43:00 drwxr-xr-x  4 root root  4096 Jan 24 04:30 3.16.3
2024-01-23 23:43:00 -rw-r--r--  1 root root   623 Jan 24 04:30 CMakeDirectoryInformation.cmake
2024-01-23 23:43:00 -rw-r--r--  1 root root 43978 Jan 24 04:30 CMakeOutput.log
2024-01-23 23:43:00 drwxr-xr-x  2 root root  4096 Jan 24 04:30 CMakeTmp
2024-01-23 23:43:00 drwxr-xr-x  3 root root  4096 Jan 24 04:30 Export
2024-01-23 23:43:00 -rw-r--r--  1 root root  7624 Jan 24 04:30 Makefile.cmake
2024-01-23 23:43:00 -rw-r--r--  1 root root 10865 Jan 24 04:30 Makefile2
2024-01-23 23:43:00 -rw-r--r--  1 root root   631 Jan 24 04:30 TargetDirectories.txt
2024-01-23 23:43:00 -rw-r--r--  1 root root    85 Jan 24 04:30 cmake.check_cache
2024-01-23 23:43:00 -rw-r--r--  1 root root     4 Jan 24 04:30 progress.marks
2024-01-23 23:43:00 drwxr-xr-x  3 root root  4096 Jan 24 04:31 sick_generic_caller.dir
2024-01-23 23:43:00 drwxr-xr-x  4 root root  4096 Jan 24 04:31 sick_scan_emulator.dir
2024-01-23 23:43:00 drwxr-xr-x  4 root root  4096 Jan 24 04:30 sick_scan_xd_api_test.dir
2024-01-23 23:43:00 drwxr-xr-x  4 root root  4096 Jan 24 04:30 sick_scan_xd_lib.dir
2024-01-23 23:43:00 drwxr-xr-x  4 root root  4096 Jan 24 04:30 sick_scan_xd_shared_lib.dir
2024-01-23 23:43:00 drwxr-xr-x  4 root root  4096 Jan 24 04:31 test_server.dir
2024-01-23 23:43:00 /bin/bash: ./run_linux_api_test_lms5xx.bash: /bin/bash^M: bad interpreter: No such file or directory

Anyone know how to help?

rostest, 2024-01-24:
Thanks for your feedback. The error `/bin/bash^M: bad interpreter` occures when Linux tries to run a bash-script with Windows-CR/LF. This can happen e.g. when github sources are cloned under Windows and then copied to Linux (and git tries to adapt CR/LF to the target).

To avoid mismatches with Linux/Windows-linefeeds, convert all CR/LF to Linux with `sudo dos2unix sick_scan_xd/test/scripts/*.bash ; sudo chmod a+x sick_scan_xd/test/scripts/*.bash` before running sick_scan_xd in docker.

AlecLozano, 2024-01-25:
It worked, thank you!

closed 2024-02-01 by rostest