SICKAG/sick_scan_xd #263: Cannot connect to Tim781 from raspberry pi when running the driver

AlecLozano, 2024-01-26:
Here are the specs of my pi:
-  running Ubuntu 20.04 with ROS2 Foxy. 
- The raspberry pi is connected to my wifi and it has an ip address of `192.168.4.70`.
- The pi is connected to the Tim781 via ethernet

Here is what I have on my Tim781:
- I changed the Tim781's ip address to `192.168.4.20`
- I changed the sensor's cola dialect from ASCII to Binary due to the errors shown

Here are the errors and warnings that show up when I run the command `ros2 launch sick_scan_xd sick_tim_7xx.launch.py hostname:=192.168.4.20` on my pi:

- `Tcp::open: Failed to open TCP Connection to 192.168.4.20, aborting`

- `Tcp::write: Connection is not open`

- `## ERROR in sendSOPASCommand(): sendCommandBuffer failed`

- `##ERROR SickScanCommon: sendSopasAndCheckAnswer("sRN SCdevicestate") failed`
- `checkColaDialect: no lidar response in any cola configuration, check lidar network!`
- `SickScanCommon:: init_scanner(): checkColaDialect failed, restarting`
- `Failed to init scanner Error Code: 1.`
- These errors repeat as the pi attempts to connect to the lidar until the process ends with: `process has died [pid 2782, exit code -9, cmd '/home/ubuntu/sick_scan_ws/ .... /sick_tim_7xx.launch hostname:=192.168.4.20 --ros-args`

When I try to ping my lidars ip address `ping 192.168.4.20` it returns `PING 192.168.4.70 icmp=1 Destination Host Unreachable`.   The ping command returns that error both when the Lidar is connected to my pi and when the Lidar is disconnected, so I think that despite the pi and the sensor being connected via ethernet with matching Ip Adresses, there is no connection being made at all.

Any help? I feel lost.

rostest, 2024-01-26:
Thanks for your error report. If `ping 192.168.4.20` fails, the lidar can not be reached due to network errors or IP configuration issues. To check the network setup, please connect the lidar to a PC in the same network, start SOPAS ET and check the IP adress, TCP port and lidar status in SOPAS ET.

If you change the IP adress of the lidar using SOPAS ET, make sure the new settings are written to the lidars EEPROM (otherwise settings will be lost). After any changes, make sure that IP adress and TCP port are identical in the lidar configuration displayed in SOPAS ET and in the sick_scan_xd launchfile.

AlecLozano, 2024-01-27:
I wrote and saved my changes to the sensors EEPROM and also changed the Ip and TCP port on the sick_scan_xd launch file. Unfortunately I am still getting the same errors when running the ping command and launching sick_scan_xd.

To test if there is any connection being made at all, I connected the Lidar's ethernet to my windows laptop (which is connected to the same network as the pi) and pinged the Lidar's ip: `ping 192.168.4.20`.  To my surprise, I got a response from the Lidar: `Reply from 192.168.4.20: bytes=32 time<1ms TTL=64`. So the problem here isn't the sensor, but the Pi. However I am confused because when I enter commands that show the ip address of my pi, such as `ifconfig` or `ip addr show`, I get the same ip: `192.168.4.70`.

closed 2024-04-08 by rostest