SICKAG/sick_scan_xd #268: Sending ColaMsg command hangs and kill the node

depauwjimmy, 2024-02-01:
I am trying sending Cola command in ROS1 using ColaMsg service.
Trying the command line first using what i could find inside documentation always end's up failing.

I start the node with roslaunch, i am using a TIM-7xx
Reading cloudpoint from /cloud works fine.

Now i want to try running :

`rosservice call /sick_lms_5xx/ColaMsg "{request: 'sMN IsSystemReady'}"`

The command hangs and on the node i see 

```
[ INFO] [1706799623.491580613]: Sending  : <STX><STX><STX><STX><Len=0017>sMN IsSystemReady CRC:<0x04>
[ WARN] [1706799623.493571785]: ## WARNING in SickScanCommon::loopOnce(): 14 byte message ignored (\x02\x02\x02\x02\x00\x00\x00\x05sFA\x00\x02v)
```

Then after a long time, the node shutdowns

```
[ERROR] [1706799773.478088105]: ## ERROR PointCloudMonitor: last point cloud message on topic "q_cloud" received 150.012 seconds ago, 150 seconds timeout exceeded.
[ERROR] [1706799773.478292284]: ## ERROR PointCloudMonitor: killing process by command "nohup sleep 1 ; kill -9 222392" for restart
nohup: ignoring input and appending output to 'nohup.out'
================================================================================REQUIRED process [q_sick-1] has died!
process has died [pid 222392, exit code -9, cmd /opt/ros/sick_ws/install/lib/sick_scan_xd/sick_generic_caller __name:=q_sick __log:=/home/god_kane/.ros/log/f4191a10-c107-11ee-ab65-73c2fece4dee/q_sick-1.log].
log file: /home/god_kane/.ros/log/f4191a10-c107-11ee-ab65-73c2fece4dee/q_sick-1*.log
Initiating shutdown!
================================================================================
[q_sick-1] killing on exit
```

Command side i receive

`ERROR: transport error completing service call: unable to receive data from sender, check sender's logs for details`

I have tried several commands and all behave the same way.
Not sure what is wrong.

rostest, 2024-02-02:
Thanks for your feedback. The lidar has answered  the ColaMsg with error code 0x02 (`sFA\x00\x02`), meaning that the lidar does not support SOPAS command "sMN IsSystemReady" (error code 2: Sopas_Error_METHODIN_UNKNOWNINDEX, Trying to access a method with an unknown Sopas index). Use SOPAS command "sRN SCdevicestate" to read the current device status: 
`rosservice call /sick_tim_7xx/ColaMsg "{request: 'sRN SCdevicestate'}"`.
 See the TiM-7xx telegram listing for a list of supported SOPAS commands.

rostest, 2024-02-05:
The sick_scan_xd driver monitors itself. If the lidar is in measurement mode, but there are no ros messages within the last 150 seconds, sick_scan_xd terminates itself with timeout error. This is intended to enable a restart after error, e.g. by script or by respawn="true" (default setting in sick_tim_7xx.launch is false). In this case, the unexpected sopas response causes the service call to fail.