SICKAG/sick_scan_xd #270: picoScan API related questions

HCHChen, 2024-02-02:
Hello,

1. When using the SickScanApiWaitNextPolarPointCloudMsg API, the returned value (ret) is consistently 5 (timeout), as shown in the image below.
[image]
[image]

2. Is there a method to use a callback function to determine if the scan has completed transmitting the entire cycle of data?
[image]

3. Is there a way to disable API packet logging in the build definition?

Best Regards,
-Henry

rostest, 2024-02-02:
Thanks for your feedback. SickScanApiWaitNextPolarPointCloudMsg should return with SICK_SCAN_API_SUCCESS, unless the lidar did not repond within the given timeout. Did you call SickScanApiWaitNextPolarPointCloudMsg with a value `message_timeout` > 0?

We recommend to register a callback via function `SickScanApiRegisterPolarPointCloudMsg` instead of calling one of the WaitNext-functions in a loop. See [Minimalistic usage example in Python](https://github.com/SICKAG/sick_scan_xd/blob/master/doc/sick_scan_api/sick_scan_api.md#minimalistic-usage-example-in-python) and [Complete usage example in Python](https://github.com/SICKAG/sick_scan_xd/blob/master/doc/sick_scan_api/sick_scan_api.md#complete-usage-example-in-python)
for a usage example.

Currently the INFO-messages from MsgPack/Compact-Exporter can not be disabled by build definition. We will add an API-function to set the log level.

rostest, 2024-02-06:
We have added API-functions SickScanApiSetVerboseLevel and SickScanApiGetVerboseLevel to set resp. get the verbose level in branch feature/api_logging on https://github.com/SICKAG/sick_scan_xd/tree/feature/api_logging .

SickScanApiSetVerboseLevel and SickScanApiGetVerboseLevel sets resp. returns the verbose level. The verbose level can be 0=DEBUG, 1=INFO, 2=WARN, 3=ERROR, 4=FATAL or 5=QUIET (equivalent to ros::console::levels). Default verbose level is 1 (INFO), i.e. sick_scan_xd prints informational, warnings and error messages on the console. Logging callbacks registered with SickScanApiRegisterLogMsg will receive all informational, warnings and error messages independant of the verbose level. Call SickScanApiSetVerboseLevel(sick_scan_library, api_handle, 2) to disable sick_scan_xd info messages. See https://github.com/SICKAG/sick_scan_xd/blob/feature/api_logging/doc/sick_scan_api/sick_scan_api.md#diagnostic for details.

Note that the WaitNext-functions of the API just return the next received message. For multiScan and picoScan, this can be a scan segment (i.e. a part of the full scan) or a fullframe poincloud (i.e. all scan points of a 360 degree scan). Depending on the timing, you may not receive all messages, i.e. you may e.g. receive scan points of different segments. We therefore recommend to register a message callback instead of a WaitNext-function. With a registered message callback, you will get all fullframe and segment pointcloud messages.

closed 2024-04-08 by rostest