SICKAG/sick_scan_xd #273: ROS services not available - multiScan (The passed service type is invalid)

boehrluSICKAG, 2024-02-06:
Hello,

we ran into an issue while we were trying to use the ROS services on our multiScan. We did everything according to the documentation and even tried different spelling of the commands, see the attached screenshot:

[image]

Since the error always shows "the passed service type is invalid" we thought something may be wrong in our configuration or setup. Therefore we checked the service list, but everything we would expect did appear in this list. So right know we are confused why these services are not working. Or are we missing a step?

[image]

Thanks in advance for pointing us in the correct direction!

Best regards,
Lukas

rostest, 2024-02-06:
Thanks for your feedback. ROS services like ColaMsg are provided by package sick_scan_xd. Therefor you have to call `source ./install/setup.bash` before running ros2 services. Otherwise ROS can not determine the service type and its data structure.

Example:
```
# ROS environment unknown:
> ros2 service type /ColaMsg
ros2: command not found
> ros2 service call /ColaMsg sick_scan_xd/srv/ColaMsgSrv "{request: 'sMN IsSystemReady'}"
ros2: command not found

# ROS environment set, but not package sick_scan_xd:
> source /opt/ros/humble/setup.bash 
> ros2 service type /ColaMsg
sick_scan_xd/srv/ColaMsgSrv
> ros2 service call /ColaMsg sick_scan_xd/srv/ColaMsgSrv "{request: 'sMN IsSystemReady'}"
The passed service type is invalid

# ROS environment set for package sick_scan_xd:
> source ./install/setup.bash
> ros2 service type /ColaMsg
sick_scan_xd/srv/ColaMsgSrv
> ros2 service call /ColaMsg sick_scan_xd/srv/ColaMsgSrv "{request: 'sMN IsSystemReady'}"
waiting for service to become available...
requester: making request: sick_scan_xd.srv.ColaMsgSrv_Request(request='sMN IsSystemReady')
response:
sick_scan_xd.srv.ColaMsgSrv_Response(response='sAN IsSystemReady 1')
```

See the telegram listing for the details of supported SOPAS commands.

boehrluSICKAG, 2024-02-15:
Hi,

thanks for your quick response, as a follow-up I tried your recommended setup steps on different machines, but I still encounter the same issue.
Please see the attached console log:
[image]

Sourcing of ROS and also the sick_scan_xd does not change the situation apparently. As you can see in my initial post the "ros2 service list" even acknowledges the different services, but I can not use them. 

_Edit: After sourcing ROS2 and sick_scan_xd I could get a result after changing the command in the following way:_
**Unsuccessfull**:
`ros2 service call /ColaMsg sick_scan_xd/srv/ColaMsgSrv "{request: 'sRN SCdevicestate'}"`

**Successfull**:
`ros2 service call /ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sRN SCdevicestate'}"`

After changing "sick_scan_xd" to "sick_scan" everything worked as expected. What's the differentiator between both variants? Different sick_scan_xd versions? How do I know which one to use?

Thanks again for your answer,
Lukas

rostest, 2024-02-16:
Thanks for following up. The package name "sick_scan" was used in driver versions 2.x. It has been replaced by "sick_scan_xd" in versions 3.x.

It looks like you are using an older version 2.x. Please update to the latest release 3.2 and rebuild. Remove all previous build files by `rm -rf ./build ./devel ./install ./build_isolated ./devel_isolated ./install_isolated` to ensure a complete rebuild. After update, use package name "sick_scan_xd" .

closed 2024-04-08 by rostest