SICKAG/sick_scan_xd #274: Some questions about SICKSegmentAPI

SICK-AliciaTee, 2024-02-08:
Hi, we have a customer Built Robotics that is trying to retrieve point cloud from picoScan using Python, the customer has installed both the sick_scan_xd drivers and sick_scan_api python module.

The following is the customer setup information:
I have successfully built and installed both the sick_scan_xd library and the sick_scan_api python module.
I have written what I think should be a functional python driver for the picoScan. I connect to the picoScan with the following launchfile args: "/path/to/sick_scan_xd/launch/sick_picoscan.launch hostname:=192.168.0.1 port:=2115 udp_receiver_ip:=192.168.0.17", where 192.168.0.17 is the static IP of my development computer.
As soon as I call the sick_scan_api.SickScanApiLoadLibrary, sick_scan_api.SickScanApiCreate, and sick_scan_api.SickScanApiInitByLaunchfile functions, rapid text begins to print on the screen. I am seeing printouts like the following at a rate that must be faster than 10 hz
[Info]: MsgPack/Compact-Exporter: 0 udp packages still in input fifo, 0 messages still in output fifo, current segment index: 1
[Info]: MsgPack/Compact-Exporter: 64600 udp scandata messages received, 64600 messages exported (scan+imu), 0% package lost.
[Info]: MsgPack/Compact-Exporter: max. 2 udp messages buffered, max 2 export messages buffered.
[Info]: MsgPack/Compact-Exporter: 64600 messages exported at 150.065 Hz, mean time: 0.879 milliseconds/messages, stddev time: 0.819, max time: 5.607 milliseconds between udp receive and messages export, histogram=[58131,1148,328,4789,146,58,0,0,0,0,0]
I am unable to retrieve polar pointclouds from the device.
When I use the sick_scan_api.SickScanApiWaitNextPolarPointCloudMsg function, it immediately returns with a timeout error code, regardless of the actual timeout that I supply to it
When I register a polar point cloud callback with the sick_scan_api.SickScanApiRegisterPolarPointCloudMsg function, the callback is never called

The customer eventually managed to retrieve the point cloud (the issue being that the customer added a port argument in the launchfile args

The customer has the following questions:
What is the reason that I am able to use the callback-based infrastructure, but using the SickScanApiWaitNextPolarPointCloudMsg function immediately times out, regardless of timeout argument?
I am still struggling to use this driver because of the significant amount of INFO printing. Is there a way that I can suppress all of the logs from the library, or at least suppress INFO logs?
What is the "reflector bit" that is alluded to in the picoscan launchfile?
What is the difference between the "echos" parameter of a custom poincloud and the host_FREchoFilter param in the launchfile? How should they be used together?
What do all of the ros-related settings (frame_id, topic names, node) do in this launchfile, given that I am running on native linux?
How can I set scan resolution and frequency to 0.05 ° and 15 Hz?
How can I configure a moving average filter, particle filter, and sensitivity, as they are made available to me in the SOPASair tool?
Kindly help in answering the above as I haven't managed to get my own Sick_scan_api working yet.

Thank you.

Best Regards,
Alicia Tee

SICK-AliciaTee, 2024-02-08:
Customer has further questions:
How can I register a callback for a custom pointcloud that includes customized pointcloud fields (coordinateNotation=3: customized pointcloud fields)? There only seems to be callback infrastructure for polar and cartesian pointclouds.
I am also keen on getting the SickScanApiWaitNext* functions working, but they still do not work.
Following the mold of the examples at, e.g. [https://github.com/SICKAG/sick_scan_xd/blob/master/python/api/sick_scan_api.py#L108](https://urldefense.com/v3/__https://github.com/SICKAG/sick_scan_xd/blob/master/python/api/sick_scan_api.py*L108__;Iw!!BXYwEpu2!tPyp76pttxzFu2qgmGnu-nvkBAiEkaB4O7n6yr9duTaAcSHI7jdU7tALMbg5Uzk337enp81eDrqbz4Sm$), I call SickScanApiClose, SickScanApiRelease, and SickScanApiUnloadLibrary in order to clean up after using the library and running the LiDAR, but this does not seem to stop the LiDAR -- I still see constant info messages.
When I am running a program that loads in this library, I need to send 2 SIGINT signals in order to exit out of the program, which results in Aborted (core dumped). Is there a cleaner/more preferred method of exiting?

rostest, 2024-02-09:
Thanks for your feedback. In general, the use of SickScanApiWaitNext-functions to receive multiScan or picoScan point clouds is not recommended. The WaitNext-functions of the API just return the next received message. For multiScan and picoScan, this can be a scan segment (i.e. a part of the full point cloud) or a fullframe scan (i.e. all scan points of a 360 degree scan). Depending on the timing, your customer may not receive all messages, which can result in incomplete point clouds. We therefore recommend to register a message callback instead of a WaitNext-function. Using a registered message callback, your customer will get all fullframe and segment point cloud messages. We recommend your customer to exchange SickScanApiWaitNext-functions by SickScanApiRegisterPointCloudMsg-functions (i.e. SickScanApiRegisterCartesianPointCloudMsg or SickScanApiRegisterPolarPointCloudMsg). See https://github.com/SICKAG/sick_scan_xd/issues/270 and https://github.com/SICKAG/sick_scan_xd/blob/feature/api_logging/doc/sick_scan_api/sick_scan_api.md#usage-example for details and examples.

> I am unable to retrieve polar pointclouds from the device.
> When I use the sick_scan_api.SickScanApiWaitNextPolarPointCloudMsg function, it immediately returns with a timeout error code, regardless [...]

I can not reproduce this error. Since sick_scan_xd receives scan data from the lidar, point clouds should be provided via API, except for possible errors during initialization. If the initialization has not been completed succesfully, the API export thread may not have been started. Were any error messages issued during initialisation? 

For further analyzation of this error, please run the following steps:

1. Checkout the latest update in branch feature/api_logging on https://github.com/SICKAG/sick_scan_xd/tree/feature/api_logging. This branch provides a cumulative update including API functions for logging and corrections of API close.

2. Rebuild sick_scan_xd as described in https://github.com/SICKAG/sick_scan_xd/blob/feature/api_logging/INSTALL-GENERIC.md#build-on-linux-generic-without-ros:
   ```
   git clone -b feature/api_logging https://github.com/SICKAG/sick_scan_xd.git
   mkdir -p ./build
   pushd ./build
   rm -rf ./*
   export ROS_VERSION=0
   cmake -DROS_VERSION=0 -DLDMRS=0 -G "Unix Makefiles" ../sick_scan_xd
   make -j4
   sudo make -j4 install    
   popd
   ```

3. Start the lidar and run the python example `sick_scan_xd_api_test.py`:
   ```
   export LD_LIBRARY_PATH=.:./build:$LD_LIBRARY_PATH
   export PYTHONPATH=.:./python/api:$PYTHONPATH
   python3 ./test/python/sick_scan_xd_api/sick_scan_xd_api_test.py ./launch/sick_multiscan.launch hostname:=192.168.0.1 udp_receiver_ip:=192.168.0.17
   ```
   sick_scan_xd_api_test.py should print `pySickScanCartesianPointCloudMsgCallback ...`, `pySickScanPolarPointCloudMsgCallback ...` and `pySickScanLogMsgCallback ...` messages. Please send a complete logfile, if error messages are printed or if Cartesian- or PolarPointCloud messages are missing.

> What is the reason that I am able to use the callback-based infrastructure, but using the SickScanApiWaitNextPolarPointCloudMsg function immediately times out, regardless of timeout argument?

I suspect the initialization was not completed or not completely successful. Please use the latest update in branch feature/api_logging and run example sick_scan_xd_api_test.py. Please post a complete logfile in case of errors or missing point cloud messages.

> I am still struggling to use this driver because of the significant amount of INFO printing. Is there a way that I can suppress all of the logs from the library, or at least suppress INFO logs?

The latest update in branch feature/api_logging provides API functions to set the verbosity level. After calling `SickScanApiSetVerboseLevel(2)`, no informational messages will be printed (warnings and errors only). See https://github.com/SICKAG/sick_scan_xd/blob/feature/api_logging/doc/sick_scan_api/sick_scan_api.md#diagnostic for details.

> What is the "reflector bit" that is alluded to in the picoscan launchfile?

The reflector bit is provided by the lidar to indicate a possible reflector. If the reflector bit is set, it can be assumed that the last echo measured originates from a reflector. On ROS, customized point clouds can be filtered by this flag.

> What is the difference between the "echos" parameter of a custom poincloud and the host_FREchoFilter param in the launchfile? How should they be used together?

On ROS, customized point clouds can be filtered by their echo index. This requires that multiple echos are activated, i.e. the picoScan echo filter is disabled by launchfile parameter host_FREchoFilter = 1 (ALL_ECHOS, EchoCount=3). We recommend to use the last echo only, which is the default setting (host_FREchoFilter = 2, LAST_ECHO, EchoCount=1)

> What do all of the ros-related settings (frame_id, topic names, node) do in this launchfile, given that I am running on native linux?

The launchfiles configure sick_scan_xd for a given lidar family independant of the underlying framework or operating system. The launchfiles are therfor identical on all platforms. ROS specific settings are not applicable on native Linux or Windows. We recommend to use the default settings.

> How can I set scan resolution and frequency to 0.05° and 15 Hz?

[... cut ...]  polar and cartesian pointclouds.

This is correct, the API currently exports only polar and cartesian point cloud messages, but not customized point clouds.

> I call SickScanApiClose, SickScanApiRelease, and SickScanApiUnloadLibrary in order to clean up after using the library and running the LiDAR, but this does not seem to stop the LiDAR. I still see [...]

Please use the latest update in branch feature/api_logging, which contains corrections for a clean exit including the stop of the lidar.

closed 2024-04-08 by rostest

rostest, 2025-11-08:
Thanks for reporting the issue.  
We need a little bit more information to investigate the problem.  
For your convenience, you’ll find a small template below.  
It would be great if you could roughly follow this structure and send us your feedback.

---

### Bug Report for sick_scan_xd (short version)

**Description**  
Briefly describe the issue and what you expected to happen.

---

**Reproduction**  
- Launch file or command used:  
  (e.g. `roslaunch sick_scan_xd test_lms1xx.launch`)  
- Steps to reproduce:  
  1. …
  2. …

Please attach the launch file and startup log if possible.

---

**Hardware / Environment**
| Parameter | Value |
|------------|-------|
| **Device type** | (e.g. LMS100, MRS6000, picoScan150) |
| **Firmware version** | |
| **Client Hardware / OS version** | (e.g. AMD386-PC with Ubuntu 24.04, Windows 11) |
| **sick_scan_xd version** | |

---

**Logs / Output**  
Paste or attach relevant log excerpts here (console, ROS, or API output).

---

**Additional context (optional)**  
Any other information that might help.

---

**Checklist**
- [ ] Attached launch file and startup log  
- [ ] Included hardware and version info  
- [ ] Tested with latest `sick_scan_xd`