SICKAG/sick_scan_xd #276: Unable to build driver using ROS 2

dillon-kong, 2024-02-12:
Hi, I am having problems building the sick driver. My end goal is to send data from the MultiScan 100 to python code. 

Hardware/firmware Overview:
- Intel Laptop 8 gb ram 4 cores
- Ubuntu 20.04 
- ROS 2 Foxy

I have attached the terminal log as a text file. 

Thank you,
Dillon Kong

[Log - Feb 12 - Log.txt](https://github.com/SICKAG/sick_scan_xd/files/14251796/Log.-.Feb.12.-.Log.txt)

Liamazing, 2024-02-12:
I'm also unable to build the driver. I've created a MRE of the failure in a Dockerfile:
```
FROM ubuntu:20.04
ENV DEBIAN_FRONTEND=noninteractive
RUN apt-get update && apt-get install -y build-essential cmake git
RUN cd /tmp/&& git clone https://github.com/SICKAG/msgpack11.git && git clone https://github.com/SICKAG/sick_scan_xd.git
RUN cd /tmp/msgpack11 && mkdir build && cd build && cmake -G "Unix Makefiles" -DMSGPACK11_BUILD_TESTS=0 -DCMAKE_POSITION_INDEPENDENT_CODE=ON .. && make -j4 && make install
RUN cd /tmp/sick_scan_xd && mkdir build && cd build && rm -rf ./* && export ROS_VERSION=0 && cmake -DROS_VERSION=0 -DLDMRS=0 -G "Unix Makefiles" .. && make -j4 && make -j4 install
```
Error output is:
```
75.18 /usr/bin/ld: libsick_scan_xd_lib.a(sick_generic_laser.cpp.o): in function `std::thread::thread<void (GenericLaserCallable::*)(), GenericLaserCallable*, void>(void (GenericLaserCallable::*&&)(), GenericLaserCallable*&&)':
75.18 /usr/include/c++/9/thread:130: undefined reference to `pthread_create'
75.18 /usr/bin/ld: libsick_scan_xd_lib.a(tcp.cpp.o): in function `std::thread::thread<void* (*)(void*), ThreadWrapperBase*, void>(void* (*&&)(void*), ThreadWrapperBase*&&)':
75.18 /usr/include/c++/9/thread:130: undefined reference to `pthread_create'
75.18 /usr/bin/ld: libsick_scan_xd_lib.a(scansegment_threads.cpp.o): in function `std::thread::thread<bool (sick_scansegment_xd::MsgPackThreads::*)(), sick_scansegment_xd::MsgPackThreads*, void>(bool (sick_scansegment_xd::MsgPackThreads::*&&)(), sick_scansegment_xd::MsgPackThreads*&&)':
75.18 /usr/include/c++/9/thread:130: undefined reference to `pthread_create'
75.18 /usr/bin/ld: libsick_scan_xd_lib.a(udp_receiver.cpp.o): in function `std::thread::thread<bool (sick_scansegment_xd::UdpReceiver::*)(), sick_scansegment_xd::UdpReceiver*, void>(bool (sick_scansegment_xd::UdpReceiver::*&&)(), sick_scansegment_xd::UdpReceiver*&&)':
75.18 /usr/include/c++/9/thread:130: undefined reference to `pthread_create'
75.18 /usr/bin/ld: libsick_scan_xd_lib.a(msgpack_converter.cpp.o): in function `std::thread::thread<bool (sick_scansegment_xd::MsgPackConverter::*)(), sick_scansegment_xd::MsgPackConverter*, void>(bool (sick_scansegment_xd::MsgPackConverter::*&&)(), sick_scansegment_xd::MsgPackConverter*&&)':
75.18 /usr/include/c++/9/thread:130: undefined reference to `pthread_create'
75.18 /usr/bin/ld: libsick_scan_xd_lib.a(msgpack_exporter.cpp.o):/usr/include/c++/9/thread:130: more undefined references to `pthread_create' follow
75.25 collect2: error: ld returned 1 exit status
75.25 make[2]: *** [CMakeFiles/sick_generic_caller.dir/build.make:85: sick_generic_caller] Error 1
75.25 make[1]: *** [CMakeFiles/Makefile2:136: CMakeFiles/sick_generic_caller.dir/all] Error 2
75.25 make: *** [Makefile:130: all] Error 2
```

rostest, 2024-02-13:
@dillon-kong Thanks for your feedback. The error messages `bash: /opt/ros/foxy/setup.bash: No such file or directory` and the colcon warnings indicate an incomplete ROS2 foxy installation. We recommend to install ROS2 foxy system-wide using prebuilt binaries instead of building ROS2 from sources within a workspace in the home directory. Try `sudo apt install ros-foxy-desktop`. After successful installation, call `source /opt/ros/foxy/setup.bash` before building sick_scan_xd.

rostest, 2024-02-13:
@Liamazing Thanks for reporting this error. A flag for linking pthread is missing in when building with option `-DLDMRS=0`. Please checkout the cumulative update in the branch feature/issue_update in https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update and rebuild.

closed 2024-04-08 by rostest