SICKAG/sick_scan_xd #279: Set scan frequency and angle resolution error (sick lms5xx)

XinyuC, 2024-02-15:
We are using a Sick LMS 511 Lidar with the latest sick_scan_xd driver. The Lidar was working properly when I used the default launch file sick_lms_5xx.launch under the launch folder. However, when I tried to change the scan frequency and angle resolution, it threw me the following error:

**Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX>sMN mLMPsetscancfg \×00\×00\×09\xc4\x00\x01\x00\x00\x06\x83...received answer: "sFA 11", expected patterns: "sAN mLMPsetscancfg"," sMA mLMPsetscancfg"**

I didn't change anything but added the following:

<param name="scan_freq" type="double" value="25" />
<param name="ang_res" type="double" value="0.1667" />

I also tried other combinations but got the same error message. Did I make any mistake here and why I got this error? I am looking forward to your reply, thanks!

rostest, 2024-02-16:
Thanks for your feedback. Not all combinations of scan frequencies and angle resolutions are valid. Which values of the parameters "ang_res" and "scan_freq" did you use? Can you post your launchfile?

You can use SOPAS ET to configure and test scan frequencies, angle resolutions and their combinations. After that, you can either save these settings in the lidars EEPROM and use the default launchfile, or you can set these values in sick_lms_5xx.launch.

XinyuC, 2024-02-16:
I set the scan_freq to 25, and ang_res to 0.1667, and left other parameters unchanged. The attached file is the launch file I used, to upload, I changed the file ext to .txt.

[sick_lms_5xx.txt](https://github.com/SICKAG/sick_scan_xd/files/14313966/sick_lms_5xx.txt)

rostest, 2024-02-19:
Thanks for following up. Your launchfile looks correct, we will test and analyze the error and come back with more details.

rostest, 2024-03-01:
Thanks for following up. Unfortunately, I can not reproduce this error. sick_scan_xd and LMS-511 initialized and worked successfully with launchfile parameters `<param name="ang_res" type="double" value="0.1667" />` and `<param name="scan_freq" type="double" value="25" />`:

```
[ INFO] [1709283644.216906106]: Sending mLMPsetscancfg request: { scan_frequency=2500, active_sector_cnt=1, angular_resolution[0]=1667, start_angle[0]=-50000, stop_angle[0]=1850000 }
[ INFO] [1709283644.216965428]: Sending  : <STX><STX><STX><STX><Len=0037>sMN mLMPsetscancfg 0x00 0x00 0x09 0xc4 0x00 0x01 0x00 0x00 0x06 0x83 0xff 0xff 0x3c 0xb0 0x00 0x1c 0x3a 0x90 CRC:<0x20>
[ INFO] [1709283644.398555432]: Receiving: <STX>sAN mLMPsetscancfg \x00\x00\x00\x09\xc4\x00\x01\x00\x00\x06\x83\xff\xff\x3c\xb0\x00\x1c\x3a...
[ INFO] [1709283644.398672522]: LMPscancfg reply: "sAN mLMPsetscancfg \x00\x00\x00\x09\xc4\x00\x01\x00\x00\x06\x83\xff\xff\x3c\xb0\x00\x1c\x3a\x90"
[ INFO] [1709283644.398739465]: LMPscancfg: { scan_frequency=2500, active_sector_cnt=1, angular_resolution[0]=1667, start_angle[0]=-50000, stop_angle[0]=1850000 }
[ INFO] [1709283644.398802610]: sAN mLMPsetscancfg: scan frequency = 25 Hz, angular resolution = 0.1667 deg.
[ INFO] [1709283644.398873117]: Sending  : <STX><STX><STX><STX><Len=0014>sRN LMPscancfg CRC:<0x63>
[ INFO] [1709283644.399893991]: Receiving: <STX>sRA LMPscancfg \x00\x00\x09\xc4\x00\x01\x00\x00\x06\x83\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX...
[ INFO] [1709283644.399976287]: LMPscancfg reply: "sRA LMPscancfg \x00\x00\x09\xc4\x00\x01\x00\x00\x06\x83\xff\xff\x3c\xb0\x00\x1c\x3a\x90"
[ INFO] [1709283644.400037053]: LMPscancfg: { scan_frequency=2500, active_sector_cnt=1, angular_resolution[0]=1667, start_angle[0]=-50000, stop_angle[0]=1850000 }
[ INFO] [1709283644.400093436]: sRA LMPscancfg: scan frequency = 25 Hz, angular resolution = 0.1667 deg.
```

[image]

[image]

The logfile is attached for information: [lms-511.log](https://github.com/SICKAG/sick_scan_xd/files/14459092/lms-511.log)

Most likely the firmware of your LMS-511 needs to be updated. The LMS-511 used for testing has firmware version 1.80. Please use SOPAS ET to check and update the firmware version.

XinyuC, 2024-03-11:
Thank you for your response. We will update the firmware and see how it works.

XinyuC, 2024-03-19:
I set up the frequency and angle resolution in the configuration with the SOPAS Engineering Tool and ran the LiDAR with the launch file with "ang_res" and "scan_freq" commented out. Then I got a warning showing that "Data length is outside acceptable range 1-381 (706). Ingoring Scan". It looks like the length of the message exceeds the default setting, do you think it is also caused by the firmware? Our LiADR has a very old firmware version 1.27, and tried to find version 1.80 online but didn't see anything useful. I am waiting for the response from SICK to get the firmware I need.

rostest, 2024-03-20:
Thanks for your response. Please contact the SICK support for firmware updates.

closed 2024-04-08 by rostest