SICKAG/sick_scan_xd #280: Failed to open TCP connection..LMS1XX

erknl, 2024-02-16:
I can successfully ping my device. However, I am unable to perform the command 'nc -z -v -w5 $SCANNERIPADDRESS 2112'. I have submitted the outputs in the attachment. I cannot complete the startup process. I need your assistance. Thank you, and have a great day.
[hata1.odt](https://github.com/SICKAG/sick_scan_xd/files/14309241/hata1.odt)
[hata2.odt](https://github.com/SICKAG/sick_scan_xd/files/14309317/hata2.odt)

rostest, 2024-02-16:
Thanks for your feedback. A missing TCP connection can have several reasons like network errors or firewall rules. Please try SOPAS Engineering Tool (SOPAS ET) for diagnostic, network setup and configuration of the lidar.

erknl, 2024-02-19:
Thank you for your response. I've checked what you mentioned and, as a final attempt, I'm addressing the issue. I configure the IP settings of my device using a tool (sopas et) and successfully obtain an image from the device. While in online mode, I connect the lidar sensor to my Jetson device. "I can ping it, but I'm encountering problems with other operations.

rostest, 2024-02-19:
Thanks for your reply. I assume you still get the error message `Failed to open TCP connection to 169.254.223.1:2112`, i.e. sick_scan_xd running on the Jetson can not establish a tcp connection to the lidar.

Missing TCP connections can have several reasons like network errors or firewall rules. Can you open a TCP connection from the Jetson to the lidar on port 2112 e.g. by `telnet 169.254.223.1 2112`? If not, a firewall might restrict access. In this case, deactivate any firewall on the Jetson or between Jetson and your lidar.

erknl, 2024-02-19:
[image]
I might be a bit of an amateur, sorry for that, and thank you for your response. If I didn't misunderstand, I sent a visual of the process you mentioned.

rostest, 2024-02-20:
Thanks for following up. According to the logfile, sick_scan_xd tries to connect to ip address `169.254.223.1`. In the screenshot on the other hand the ip address is `192.168.6.151`. Please make sure you got the correct ip address of your lidar and run `ros2 launch sick_scan_xd sick_tim_7xx.launch.py hostname:=192.168.6.151` (resp. the lidars ip address).

erknl, 2024-02-20:
I made some adjustments after my initial post to address the issue. That's why the IP addresses are different between the posts. 

The output of the command is attached. (error 3)
[hata3.odt](https://github.com/SICKAG/sick_scan_xd/files/14343923/hata3.odt)
> ros2 launch sick_scan_xd sick_lms_1xx.launch.py ​​hostname:=192.168.6.151
Images of the sick_lms_1xx.launch file, ping and ifconfig commands are also attached.
[image]
The sopas connection settings screen on Windows PC is also attached.
[image]

rostest, 2024-02-20:
This looks very promising! 👍

closed 2024-04-08 by rostest