SICKAG/sick_scan_xd #281: Get data from several lidars simultaneously

paulinelvne, 2024-02-19:
Hello, 

After integrating the sick_scan_xd library (C++ only mode, no ROS), we realized that it wasn't possible get data from several lidars simultaneously.
We think it's important that this point appears in the sick_scan_xd documentation, because discovering this after integration was an unpleasant surprise for us.

After discovering this we figured out that we were not the only ones to have missed this point and that other peoples assumed that it was possible to use the library this way: [#269](https://github.com/SICKAG/sick_scan_xd/issues/269), [#218](https://github.com/SICKAG/sick_scan_xd/issues/218, [#194](https://github.com/SICKAG/sick_scan_xd/issues/194), [169](https://github.com/SICKAG/sick_scan_xd/issues/169).

We also work with other lidar brands, with which it's possible to connect as many lidars as we like. We often use multiple lidars in the projects we have and it would have been nice to have this feature for our use case.

Anyway, we understand that it is possible use this library with multiple lidars by starting multiple processes, each one handling a single sensor, so we will probably go down this route.

Is an implementation of sick_scan_segment_xd planned in the future that will support multiple lidars without having to deal with multiple processes on our side, without ROS ?

At least it would be nice to state in an obvious location of the doc that using multiple lidars without ROS means that we have to plan for 1 process per lidar.

Thanks in advance

rostest, 2024-02-20:
Many thanks for your feedback! We updated the documentation in sick_scan_api.md (pull request #283). We will clarify a possible API support for multiple lidars with SICK.

rostest, 2024-03-19:
Thank you for following up. Possible API support for simultaneous use of  multiple lidars is currently being clarified. Please contact SICK support for feature requests.

closed 2024-04-08 by rostest