SICKAG/sick_scan_xd #282: ROS launch runs but no data come out - multiScan

SICK-AliciaTee, 2024-02-19:
Dear Sir or Madam, 

Could you take a look at the attached log file, the customer has sick_scan_xd installed on ROS1 Noetic and Ubuntu platform, the launch file runs but no data came out, we looked at their ethernet setting and everything seems normal.

We have also tried uninstall and install the driver but still running the same issue. 

Thank you. 

![SOPAS_Air_Terminal](https://github.com/SICKAG/sick_scan_xd/assets/151696922/f
[image]
dfbd971-c10a-4447-8af3-4ce9b95f997d)
g seems normal. 
[roslaunch_terminal.txt](https://github.com/SICKAG/sick_scan_xd/files/14335589/roslaunch_terminal.txt)

Thank you, 
[image]

SICK-AliciaTee, 2024-02-21:
Hi rostest, 

We tried to change the IP address to 192.168.x.x, but still run into the same issue. 

Please see the screen shots and information, kindly advise. 
[launch_terminal.txt](https://github.com/SICKAG/sick_scan_xd/files/14366147/launch_terminal.txt)
[image]
[image]
[image]

Thank you.

rostest, 2024-02-22:
Thanks for logfiles and screenshots. Your setup looks fine. We will test and analyze the issue and come back to it.

rostest, 2024-02-29:
Unfortunately I can not reproduce this error. Please run the following commands on the customizers PC:

```
mkdir -p ./ws_sick_scan_xd_ros1/src
cd ./ws_sick_scan_xd_ros1/src
git clone https://github.com/SICKAG/sick_scan_xd.git
cd ..
source /opt/ros/noetic/setup.bash
rm -rf ./build_isolated/ ./devel_isolated/ ./install_isolated/
catkin_make_isolated --install --cmake-args -DROS_VERSION=1 -DLDMRS=0 -Wno-dev
source ./devel_isolated/setup.bash
roslaunch sick_scan_xd sick_multiscan.launch hostname:=192.168.0.1 udp_receiver_ip:=192.168.0.50
```

These are exactly the commands I used to view the multiScan pointcloud in rviz (except for a different ip address, i.e. argument `udp_receiver_ip:=192.168.0.100`):

[image]

In the attachment you find a complete logfile of my build and run for information. Please run the commands above and post a logfile.

If the problem still remains (i.e. no pointcloud in rviz or rostopic), please check for the latest firmware update. If the problem remains, please start wireshark and save a short sequence of the network traffic in a pcapng-file (some seconds are enough). With this pcapng-file we can analyze the tcp and udp communication between sick_scan_xd and the lidar. Thanks in advance!

[20240229_multiscan.zip](https://github.com/SICKAG/sick_scan_xd/files/14449584/20240229_multiscan.zip)

Heiro008, 2024-03-22:
Hi @rostest ,
I am facing the same issue in ros2 driver with multiscan100, i'm using ubuntu 22.04 with ros2 humble.
It seems like i cannot receive any scan data via UDP.
FYI firewall is disabled.
Sopas air is working fine and i could see point cloud data in browser but not in any ros topics.
I have compared your previous log file and found that my launch stopped after the following logs,
"""
[sick_generic_caller-1] [INFO] [1711083596.743137085] [sick_scan_xd]: Sending  : <STX>sMN LMCstartmeas<ETX>
[sick_generic_caller-1] [INFO] [1711083596.750921893] [sick_scan_xd]: Receiving: <STX>sAN LMCstartmeas 0<ETX>
[sick_generic_caller-1] [INFO] [1711083596.750952880] [sick_scan_xd]: SickScanServices: Request "sMN LMCstartmeas" successfully sent, received reply "sAN LMCstartmeas 0"
[sick_generic_caller-1] [INFO] [1711083596.7509615
[log.txt](https://github.com/SICKAG/sick_scan_xd/files/14717589/log.txt)
49] [sick_scan_xd]: SickScanServices::sendSopasCmdCheckResponse(): request: "sMN LMCstartmeas", response: "sAN LMCstartmeas 0"
"""
I have attached the log file and wireshark capture.
kindly help to solve this.

Thanks in advance!

[multiscan_run.zip](https://github.com/SICKAG/sick_scan_xd/files/14717678/multiscan_run.zip)

rostest, 2024-03-25:
@Heiro008 Thanks for your feedback and logfiles. Parameter "udp_receiver_ip" must be the ip address of the machine receiving the scan data from the multiScan via UDP, i.e. the ip address of your PC running sick_scan_xd. According to the logfile, parameter udp_receiver_ip:=192.168.0.100 has been used (the ip address of the previous example). Wireshark displays IP traffic between 192.168.0.1 (i.e. the lidar) and 192.168.0.5. 

192.168.0.5 is probably the ip address of your PC. Run sick_scan_xd with parameter `udp_receiver_ip:=192.168.0.5`, e.g.:
`ros2 launch sick_scan_xd sick_multiscan.launch.py hostname:=192.168.0.1 udp_receiver_ip:=192.168.0.5
`

Command `ipconfig /all` (on Windows) resp. `ifconfig -a` (on Linux) displays the ip address. Alternatively you can use the network configuration in Sopas Air to get or set the ip addresses.

Heiro008, 2024-03-25:
Got it, its working now. Thanks

closed 2024-04-08 by rostest