SICKAG/sick_scan_xd #284: How to make MultiScan support field monitoring in sick_scan_xd

henaiguo, 2024-02-27:
I am testing Multiscan136 and have already set up the field evaluation function through the web.   
[image]
However, I have found that only the LMS1xx, LMS5xx, TiM7xx, and TiM7xxS families support field evaluation.  
[field_monitoring_extensions](https://github.com/SICKAG/sick_scan_xd/blob/develop/doc/field_monitoring_extensions.md)
So I want to know if Multiscan supports field evaluation, and if so, what should I do?

rostest, 2024-02-27:
Thanks for your feedback. Field monitoring for multiScan is currently not supported by the sick_scan_xd driver. We will clarify a possible support in a future release with SICK.

henaiguo, 2024-02-27:
Thanks for your reply. That is to say, the result of receiving field monitoring from multiScan can only be processed through IO?

rostest, 2024-02-27:
@henaiguo Yes, that is correct.

closed 2024-02-28 by henaiguo