SICKAG/sick_scan_xd #285: When will the picScan100 drivers be released?

Kaju-Bubanja, 2024-02-27:
I'm trying to run a picoScan100 scanner, but I can't find the drivers, are they already available or will they be released?

rostest, 2024-02-27:
Thanks for your feedback. A prerelease of sick_scan_xd for picoScan150 Core w/o Add On (picoScan100) is available in branch [feature/api_logging](https://github.com/SICKAG/sick_scan_xd/tree/feature/api_logging).

Kaju-Bubanja, 2024-02-28:
Great, while the driver itself isn't really stable and mature yet, I got it to work and like the results, is there a roadmap for a release of the product/driver?

rostest, 2024-02-28:
Support for picoScan100 devices will be added in the next release. Can you be more specific about "the driver itself isn't really stable and mature yet"? Is there a driver bug related to the new picoScan100 or a general instability? In case of sick_scan_xd issues, please do not hesitate to provide specific details.

Kaju-Bubanja, 2024-02-28:
Starting, restarting and stopping doesn't work and I had to reboot the device a few times, the launch file looks a bit messy and I think needs some polish. I can't go trough each point here, since I'm not part of your test departement, but it's obvious that it needs more polish. But I'm excited to work with it when it comes out.

closed 2024-02-28 by Kaju-Bubanja