SICKAG/sick_scan_xd #288: ROS-PicoScan150-error too many initiallizers

sjeffreydong, 2024-03-04:
When i run "sick_generic_caller" happeded issues , what should i do next ?
[image]

[image]
[image]
[image]

rostest, 2024-03-04:
Thanks for your feedback. It looks like your compiler does not support C++ 14. Which OS and gcc version do you use?

sjeffreydong, 2024-03-04:
Dear,
Please have a look !
[image]
[image]

rostest, 2024-03-04:
Thanks for following up. Ubuntu 16 is not supported any longer (end of life). As far as I understand, full C++14 support incl. libstdc++ starts with gcc 6.1. Please upgrade your system to Ubuntu 20 or newer.

sjeffreydong, 2024-03-05:
Dear Teams,
I asked cutstomer they can't upgrade  Ubuntu to 20, because of some complex reasons, and what we can do to adapt their current version for connecting PicoScan?

rostest, 2024-03-05:
sick_scan_xd supports Ubuntu 20 or newer. Older systems are currently not supported.

closed 2024-03-05 by rostest