SICKAG/sick_scan_xd #291: Compile error on Raspberry PI4 OS

kinni318, 2024-03-08:
A customer who have to use the sick_scan_xd driver on a raspberry PI OS.
When they tried to compile the library they got a strange error:
----------
[ 96%] Building CXX object CMakeFiles/sick_scan_xd_lib.dir/roswrap/src/rossimu/k                                                                                                                                                             inetic/src/rate.cpp.o
[ 97%] Building CXX object CMakeFiles/sick_scan_xd_lib.dir/roswrap/src/rossimu/k                                                                                                                                                             inetic/src/rossimu.cpp.o
[ 98%] Building CXX object CMakeFiles/sick_scan_xd_lib.dir/roswrap/src/tools/sic                                                                                                                                                             k_scan/pointcloud_utils.cpp.o
[ 99%] Linking CXX shared library libsick_scan_xd_shared_lib.so
[ 99%] Built target sick_scan_xd_shared_lib
[ 99%] Linking CXX static library libsick_scan_xd_lib.a
[ 99%] Built target sick_scan_xd_lib
Scanning dependencies of target sick_generic_caller
[ 99%] Building CXX object CMakeFiles/sick_generic_caller.dir/driver/src/sick_ge                                                                                                                                                             neric_caller.cpp.o
[100%] Linking CXX executable sick_generic_caller
/usr/bin/ld: libsick_scan_xd_lib.a(udp_receiver.cpp.o): in function `sick_scanse                                                                                                                                                             gment_xd::UdpReceiver::Start()':
/usr/include/c++/8/bits/unique_ptr.h:81: undefined reference to `pthread_create'
/usr/bin/ld: libsick_scan_xd_lib.a(msgpack_converter.cpp.o): in function `sick_s                                                                                                                                                             cansegment_xd::MsgPackConverter::Start()':
/usr/include/c++/8/bits/unique_ptr.h:81: undefined reference to `pthread_create'
/usr/bin/ld: libsick_scan_xd_lib.a(msgpack_exporter.cpp.o): in function `sick_sc                                                                                                                                                             ansegment_xd::MsgPackExporter::Start()':
/usr/include/c++/8/bits/unique_ptr.h:81: undefined reference to `pthread_create'
/usr/bin/ld: libsick_scan_xd_lib.a(sick_generic_laser.cpp.o): in function `start                                                                                                                                                             GenericLaser(int, char**, std::__cxx11::basic_string<char, std::char_traits<char                                                                                                                                                             >, std::allocator<char> >, roswrap::NodeHandle*, int*)':
/usr/include/c++/8/ext/new_allocator.h:86: undefined reference to `pthread_creat                                                                                                                                                             e'
/usr/bin/ld: libsick_scan_xd_lib.a(tcp.cpp.o): in function `Tcp::open(std::__cxx                                                                                                                                                             11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned                                                                                                                                                              short, bool)':
/usr/include/c++/8/bits/basic_string.h:1932: undefined reference to `pthread_cre                                                                                                                                                             ate'
/usr/bin/ld: libsick_scan_xd_lib.a(scansegment_threads.cpp.o):/usr/include/c++/8                                                                                                                                                             /bits/unique_ptr.h:81: more undefined references to `pthread_create' follow
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/sick_generic_caller.dir/build.make:85: sick_generic_cal                                                                                                                                                             ler] Error 1
make[1]: *** [CMakeFiles/Makefile2:136: CMakeFiles/sick_generic_caller.dir/all]                                                                                                                                                              Error 2
make: *** [Makefile:130: all] Error 2
-------------------
They followed the build description-[Build on Linux generic without ROS] 
and used options -DLDMRS=0 -DRASPBERRY=1 to skip LDMRS sensors...
[cmake -DROS_VERSION=0 -DLDMRS=0 -DRASPBERRY=1 -G "Unix Makefiles" ../sick_scan_xd]

(https://github.com/SICKAG/sick_scan_xd/blob/develop/INSTALL-GENERIC.md#build-on-linux-generic-without-ros)

Thier environment and specifications:
   Raspberry Pi 4 Model B
   OS: Raspberry PI OS 
          whiout ROS
   LiDAR: SICK multiScan

I would appreciate it if you could tell me how to solve it.

rostest, 2024-03-08:
Thanks for your feedback! sick_scan_xd uses std::thread, therefore target sick_generic_caller needs to be linked to pthread. Depending on system settings, the linker flag is implicitly supplemented or or must be set explicitly. 

Please use the latest update (version 3.2.2) in branch https://github.com/SICKAG/sick_scan_xd/tree/feature/bloom_humble and rebuild.

kinni318, 2024-03-08:
Thanks for your answer!
I have treid on my raspberryPI OS with [-b feature/bloom_humble] and the error still there... 
=>git clone -b feature/bloom_humble https://github.com/SICKAG/sick_scan_xd.git
----------------------
sick@raspberrypi:~ $ mkdir -p ./sick_bloomH2
sick@raspberrypi:~ $ cd ./sick_bloomH2
sick@raspberrypi:~/sick_bloomH2 $ git clone https://github.com/SICKAG/libsick_ldmrs.git
Cloning into 'libsick_ldmrs'...
remote: Enumerating objects: 404, done.
remote: Counting objects: 100% (16/16), done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 404 (delta 5), reused 9 (delta 3), pack-reused 388
Receiving objects: 100% (404/404), 419.30 KiB | 973.00 KiB/s, done.
Resolving deltas: 100% (209/209), done.
sick@raspberrypi:~/sick_bloomH2 $ git clone -b feature/bloom_humble https://github.com/SICKAG/sick_scan_xd.git
Cloning into 'sick_scan_xd'...
remote: Enumerating objects: 8212, done.
remote: Counting objects: 100% (2550/2550), done.
remote: Compressing objects: 100% (609/609), done.
remote: Total 8212 (delta 2032), reused 2337 (delta 1917), pack-reused 5662
Receiving objects: 100% (8212/8212), 56.53 MiB | 1.58 MiB/s, done.
Resolving deltas: 100% (6128/6128), done.
Updating files: 100% (1400/1400), done.
sick@raspberrypi:~/sick_bloomH2 $ mkdir -p ./build
sick@raspberrypi:~/sick_bloomH2 $ pushd ./build
~/sick_bloomH2/build ~/sick_bloomH2
sick@raspberrypi:~/sick_bloomH2/build $ rm -rf ./*
sick@raspberrypi:~/sick_bloomH2/build $ export ROS_VERSION=0
sick@raspberrypi:~/sick_bloomH2/build $ cmake -DROS_VERSION=0 -DLDMRS=0 -DRASPBERRY=1 -G "Unix Makefiles" ../sick_scan_xd
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Option CMAKE_ENABLE_EMULATOR = , building sick_scan_xd without emulator support, ENABLE_EMULATOR=OFF
-- ROS_VERSION precheck: ENV{ROS_VERSION}=0 ENV{ROS_DISTRO}= ROS_VERSION=0 CMAKE_PREFIX_PATH=
-- ROS_VERSION=0
-- Building sick_scan_xd for target RASPBERRY
-- Building sick_scan_xd for raspberry without ldmrs support
-- Building sick_scan_xd with SCANSEGMENT_XD support
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sick/sick_bloomH2/build
sick@raspberrypi:~/sick_bloomH2/build $ make -j4
[  2%] Building CXX object CMakeFiles/sick_scan_xd_api_test.dir/driver/src/sick_scan_xd_api/sick_scan_api_converter.cpp.o
[  2%] Building CXX object CMakeFiles/sick_scan_xd_api_test.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_test.cpp.o
[  3%] Building CXX object CMakeFiles/sick_scan_xd_lib.dir/driver/src/abstract_parser.cpp.o
[  4%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/abstract_parser.cpp.o
[  5%] Building C object CMakeFiles/sick_scan_xd_api_test.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c.o
[  5%] Building CXX object CMakeFiles/sick_scan_xd_api_test.dir/test/src/sick_scan_xd_api/toojpeg/toojpeg.cpp.o
In file included from /home/sick/sick_bloomH2/sick_scan_xd/roswrap/src/rossimu/melodic/include/nav_msgs/Odometry.h:20,
                 from /home/sick/sick_bloomH2/sick_scan_xd/include/sick_scan/sick_ros_wrapper.h:131,
                 from /home/sick/sick_bloomH2/sick_scan_xd/include/sick_scan/abstract_parser.h:39,
                 from /home/sick/sick_bloomH2/sick_scan_xd/driver/src/abstract_parser.cpp:63:
/home/sick/sick_bloomH2/sick_scan_xd/roswrap/src/rossimu/kinetic/include/geometry_msgs/PoseWithCovariance.h:46:20: error: field ‘covariance’ has incomplete type ‘geometry_msgs::PoseWithCovariance_<ContainerAllocator>::_covariance_type’ {aka ‘std::array<double, 36>’}
   46 |   _covariance_type covariance;
      |                    ^~~~~~~~~~
In file included from /usr/include/c++/12/bits/unique_ptr.h:36,
                 from /usr/include/c++/12/memory:76,
                 from /home/sick/sick_bloomH2/sick_scan_xd/include/sick_scan/sick_ros_wrapper.h:71:
/usr/include/c++/12/tuple:1595:45: note: declaration of ‘geometry_msgs::PoseWithCovariance_<ContainerAllocator>::_covariance_type’ {aka ‘struct std::array<double, 36>’}
 1595 |   template<typename _Tp, size_t _Nm> struct array;
      |                                             ^~~~~
[... 4 lines cut ...]
/home/sick/sick_bloomH2/sick_scan_xd/roswrap/src/rossimu/kinetic/include/geometry_msgs/PoseWithCovariance.h:46:20: error: field ‘covariance’ has incomplete type ‘geometry_msgs::PoseWithCovariance_<ContainerAllocator>::_covariance_type’ {aka ‘std::array<double, 36>’}
   46 |   _covariance_type covariance;
      |                    ^~~~~~~~~~
In file included from /usr/include/c++/12/bits/unique_ptr.h:36,
                 from /usr/include/c++/12/memory:76,
                 from /home/sick/sick_bloomH2/sick_scan_xd/include/sick_scan/sick_ros_wrapper.h:71:
/usr/include/c++/12/tuple:1595:45: note: declaration of ‘geometry_msgs::PoseWithCovariance_<ContainerAllocator>::_covariance_type’ {aka ‘struct std::array<double, 36>’}
 1595 |   template<typename _Tp, size_t _Nm> struct array;
      |                                             ^~~~~
[... cut ...]
make[1]: *** Waiting for unfinished jobs....
[  9%] Building CXX object CMakeFiles/sick_scan_xd_lib.dir/driver/src/helper/angle_compensator.cpp.o
make[2]: *** [CMakeFiles/sick_scan_xd_lib.dir/build.make:76: CMakeFiles/sick_scan_xd_lib.dir/driver/src/abstract_parser.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:89: CMakeFiles/sick_scan_xd_lib.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
sick@raspberrypi:~/sick_bloomH2/build $ 
---------
T

rostest, 2024-03-08:
Thanks for following up. The build error is different, it looks like the compiler either does not fully support C++ 14, or that an additional `#include <array>` is required in `PoseWithCovariance.h`. Which linux, compiler and compiler version do you use? Can you add `#include <array>` in file `/home/sick/sick_bloomH2/sick_scan_xd/roswrap/src/rossimu/kinetic/include/geometry_msgs/PoseWithCovariance.h` for a quick test and retry?

kinni318, 2024-03-11:
Many thanks for your help. it worked. perfectlly done without error by editing the file before building the lib. 
 ->OS  info:
 Linux raspberrypi 6.1.0-rpi8-rpi-v8 / GNU 12.2.0

rostest, 2024-03-13:
Thanks for following up! We will add the missing include in the next release.

closed 2024-04-08 by rostest