SICKAG/sick_scan_xd #295: No "time" field in the PointCloud2 output using the ROS1 driver

aubin-tchoi, 2024-03-13:
Hello,

I successfully built the ROS1 driver and have everything working very well but found out that there was no field containing the point-wise timestamps in the PointCloud2 messages output.
I was wondering whether this is something that will be added in the future or can already be accessed with some minor tinkering on the driver.

Thanks in advance.

rostest, 2024-03-13:
Thanks for your feedback. Most scanner transmit one 32 bit timestamp for each scan (detail from the telegram listing):
[image]
Point-wise timestamps are currently not provided.

aubin-tchoi, 2024-03-15:
Thank you for your feedback, what I am looking for is indeed point-wise timestamps. Do you find this is something that you are planning to add in the future to the ROS driver?

rostest, 2024-03-15:
Point timestamps are not currently planned. They could be calculated in the following way:

The point cloud time stamp is the transmission time stamp, i.e. the time stamp of the last shot converted to system time. The time step `dt` in seconds between two points is constant and can be calculated as follows

`dt = (1 / scan_frequency) * (max_azimuth - min_azimuth) / (360 degrees)`

given the horizontal scan range `(max_azimuth - min_azimuth)` resp. the azimuth range of a scan segment in degree and the `scan_frequency` in Hz.

The timestamp of the ith point can be calculated as `t_i = pointcloud_timestamp - (N - 1 - i) * dt` with 0 <= i < N and N points in the point cloud.

Note that this calculation works for common lidars with unsegmented point clouds, i.e. all point clouds that are not cumulated over different scan segments. In the case of point clouds concatenated from segments, i.e. the full frame point clouds of multiScan and picoScan lidars, the points are not ordered by ascending time. In this case, the calculation must be performed for each individual segment.

closed 2024-03-18 by aubin-tchoi