SICKAG/sick_scan_xd #298: LMS511 ROS2 run problem

ajsdlasd, 2024-03-19:
Hi everyone,
When executing "ros2 run sick_scan_xd sick_generic_caller ./src/sick_scan_xd/launch/sick_lms_5xx.launch hostname:=192.168.1.233" ,
i get the following error:

[INFO] [1710832634.605771011] [sick_scan_xd]: SickScanServices: ros services initialized
[INFO] [1710832634.605788364] [sick_scan_xd]: Setup completed, sick_scan_xd is up and running. Pointcloud is published on topic "cloud"
[INFO] [1710832634.762275580] [sick_scan_xd]: Software PLL locking started, mapping ticks to system time.
[INFO] [1710832634.762389176] [sick_scan_xd]: 1 / 6 packets dropped. Software PLL not yet locked.
[ERROR] [1710832634.762821208] [sick_scan_xd]: ## ERROR SickScanMessages::parseLFErecMsg(): parse error, fields number = 0 should be greater than 0 (/home/hql/sick_scan_ws/src/sick_scan_xd/driver/src/sick_scan_messages.cpp:267)
[ERROR] [1710832634.762853481] [sick_scan_xd]: ## ERROR SickScanMessages::parseLFErecMsg(): 22 byte received: \x02\x02\x02\x02\x00\x00\x00\x0dsSN LFErec \x00\x00U
[WARN] [1710832634.762875253] [sick_scan_xd]: ## ERROR SickScanCommon: parseLFErecMsg failed, received 22 byte LFErec \x02\x02\x02\x02\x00\x00\x00\x0dsSN LFErec \x00\x00U
[INFO] [1710832634.779794501] [sick_scan_xd]: 2 / 6 packets dropped. Software PLL not yet locked.
[INFO] [1710832634.799120057] [sick_scan_xd]: 3 / 6 packets dropped. Software PLL not yet locked.
[INFO] [1710832634.819780388] [sick_scan_xd]: 4 / 6 packets dropped. Software PLL not yet locked.
[INFO] [1710832634.839059541] [sick_scan_xd]: 5 / 6 packets dropped. Software PLL not yet locked.
[INFO] [1710832634.859843299] [sick_scan_xd]: Software PLL is ready and locked now!

Could someone know what the reason and how to solve the problem?

Thank you for your answers.

rostest, 2024-03-20:
Thanks for your feedback. The field monitoring telegram "sSN LFErec \x00\x00" from the lidar does not contain a valid field number and is therefore unexpected. 

If you do not need field monitoring, or if you do not handle ros messages of type [LFErec ](https://github.com/SICKAG/sick_scan_xd/blob/develop/msg/LFErecMsg.msg), you can just deactivate LFErec field monitoring telegrams in the launchfile sick_lms_5xx.launch:
```
<param name="activate_lferec" type="bool" value="False"/> <!-- activate/deactivate field monitoring by lferec messages -->
```
Otherwise please check for firmware updates using SOPAS ET.

ajsdlasd, 2024-03-21:
> Thanks for your feedback. The field monitoring telegram "sSN LFErec \x00\x00" from the lidar does not contain a valid field number and is therefore unexpected.
> 
> If you do not need field monitoring, or if you do not handle ros messages of type [LFErec ](https://github.com/SICKAG/sick_scan_xd/blob/develop/msg/LFErecMsg.msg), you can just deactivate LFErec field monitoring telegrams in the launchfile sick_lms_5xx.launch:
> 
> ```
> <param name="activate_lferec" type="bool" value="False"/> 
> ```
> 
> Otherwise please check for firmware updates using SOPAS ET.

Thanks, when I changed the variable True to False, he really didn't report an error, but I wanted to ask if there was any effect after turning this off, or what type LFErec does. Thank you for your reply.

closed 2024-04-08 by rostest