SICKAG/sick_scan_xd #301: Not found time, ring, intensity field for mutilScan100 lidar sensor

IoTTN, 2024-03-24:
Hi, I tried to convert Sick multiScan100 MULS1AA-112211 lidar packets to pcl::PointCloud<pcl::PointXYZI> using the standard pcl::fromROSMsg. I have the error complaining about the conversion can't find the field time, intensity, ring. However, carefully inspect the sensor_msgs::PointCloud2 message, I see that it has the intensity field. I want use Point-LIO to mapping create 3D map. Please help me. Thank you.

rostest, 2024-03-25:
Thanks for your feedback. To be on the safe side, you can allocate a pcl::PointCloudpcl point cloud with (height * width) points and convert each point in the sensor_msgs::PointCloud2 message to a pcl::PointCloudpcl::PointXYZI point. See https://github.com/SICKAG/sick_scan_xd/blob/develop/doc/sick_scan_segment_xd.md for details about the memory layout.

IoTTN, 2024-03-26:
@rostest  Thank you for your feedback, I'm trying to correct it according to your comments. I am learning more about the concept of ring , time field, how can I extract more of the above fields? I will try to reflect as soon as possible on what to do next.

closed 2024-05-03 by rostest