SICKAG/sick_scan_xd #303: Running MRS_6xxx driver on RaspberryPi5 with ROS1 Noetic

contradictory, 2024-03-26:
Hi team,

I want to receive data on a Raspberry Pi5 and transmit them to server. I install Docker and run an image with Ubuntu20 and ROS1 installed, then build the sick_scan_xd file in the container. I noticed the statement that LDMRS_6xxx is not supported on Raspberry, but I still tried to build the sick_scan_xd, without setting "-DRASPBERRY=1". When I try to run the launch file, it has these errors. It cannot set up the TCP connection between lidar and Raspberry Pi. Does that mean MRS_6xxx is not supported on Raspberry Pi? 

[image]

Is there other solutions could figure this? Could you recommend better solutions for me?
Thanks so much!

rostest, 2024-03-28:
Thanks for your feedback. Except for LDMRS lidars, sick_scan_xd supports MRS-6xxx and all devices listed in https://github.com/SICKAG/sick_scan_xd/blob/develop/REQUIREMENTS.md on both x64 and ARM architectures including Raspberry. If you have a MRS-6xxx lidar, build with cmake flags `-DLDMRS=0 -DRASPBERRY=1` as described in https://github.com/SICKAG/sick_scan_xd/blob/develop/INSTALL-ROS1.md. Optional cmake-flag `-DRASPBERRY=1` activates some minor optimizations on Raspberries. Run the driver with `roslaunch sick_scan_xd sick_mrs_6xxx.launch hostname:=<mrs6xxx_ip_adress>`, where `<mrs6xxx_ip_adress>` has to be the ip adress of your lidar. 

The screenshot shows that sick_scan_xd can not open a tcp connection to `192.168.0.1`, which is the default ip adress of MRS-6xxx lidars. Make sure, that your lidar is reachable (on Raspberry and in docker) and that the correct ip address is given by parameter `hostname:=<mrs6xxx_ip_adress>`. If in doubt, use SOPAS ET to get or set the lidars ip adress and try `ping <mrs6xxx_ip_adress>` to check that the device can be reached. 

sick_scan_xd uses [libsick_ldmrs](https://github.com/SICKAG/libsick_ldmrs) to support LDMRS lidars. As far as I know, libsick_ldmrs does not support Raspberries resp. ARM architectures.  If you are using a LDMRS, you may nevertheless try by building libsick_ldmrs as decscribed in https://github.com/SICKAG/sick_scan_xd/blob/develop/INSTALL-GENERIC.md#build-on-linux-generic-without-ros and cmake flag `-DLDMRS=1`.

contradictory, 2024-03-28:
Thanks! Clarify a lot in the post. 
Now I pinged the address for lidar 192.168.0.5 and got the response in the container. And the driver still cannot set up the TCP connection. I think it is not the problem with the driver, but the problem from the docker and container. 

[image]
[image]

I am trying to rebuild sick_scan_xd. Or try a normal way without docker, building ROS2 version.

contradictory, 2024-03-28:
After adjusted IP address again, it works, and now receiving data in container! Thanks!

closed 2024-03-28 by contradictory