SICKAG/sick_scan_xd #306: How to use ROS1 to launch 2 PicoScan?

sjeffreydong, 2024-04-02:
Dear Teams,

Our customer Ros version ROS-NOETIC 20.04. they use ROS1 driver.

They want to use 2 PicoScan running on 1 PC controller，However they can run 1 picoScan on controller, but they can't run 2 picoScan at the same time, as the picture shows:

[image]

So how to run 2 picoScan on 1 controller at the same time?

Jonny0816, 2024-04-02:
Dear Sir, 
Customer is using 3.20 version and had defined the difference port for difference LiDAR in difference launch files. Customer feedback used with lower version is ok, but 3.20 version not. Please help to analysis and give a recommend about this.
[image]

[image]

rostest, 2024-04-03:
Thanks for your feedback. Running 2 picoScan lidars with 2 launchfiles in 2 ros nodes should work without problems. The screenshots of the launchfiles with different settings look good.

The messages in the first screenshot do not quite match the launchfiles. According to the launchfiles, the ros nodes should have the names "pico_scan_head" resp. "pico_scan_tail" instead of "sick_picoscan1" resp. "sick_picoscan2", which are displayed by ros. The error messages show that an empty ip address is used instead of the configured ones. It looks like the launchfiles are not used or their configuration is not active. Most likely, sick_scan_xd uses identical default values in both ros nodes instead of using the launchfiles resp. the configuration in the launchfiles.

The customer may simply have used other launchfiles with identical default settings. Can you post a complete logfile including the ros calls and complete launchfiles?

closed 2024-05-03 by rostest