SICKAG/sick_scan_xd #309: Get contamination warning and error for Lidar_MRS1000 in callback function in sick_scan_xd API

janvanz62, 2024-04-02:
We need the contamination information from the "sick Lidar sensor MRS1000" using the "sick_scan_xd" API.

Using the SopasET tool we get the warning and error meassage regarding contamination of the MRS1000 sensor, see screenshot inside attached zip file.

We get scan pointcloud data from a registered SickScanApiRegisterCartesianPointCloudMsg callback function.  
A registerd callback funtion from "SickScanApiRegisterRadarScanMsg" displaying the warning and error contamination information 
is not invoked inside the running application.

Attached is the logging of the executed "minimum_sick_scan_api_client" application and the used "sick_sensor.lauch" configuration file.
[sick_scan_xd_MRS1000.log.zip](https://github.com/SICKAG/sick_scan_xd/files/14835515/sick_scan_xd_MRS1000.log.zip)

Q1: Is there a special setting required for the MRS1000 inside the "sick_sensor.launch" file to get support the "SickScanApiRegisterRadarScanMsg" callback function? 
Q2: Do we get the contamintaion information from the MRS1000 only using telegrams from the "roslaunch" tool?

rostest, 2024-04-03:
Thanks for your feedback. API function SickScanApiRegisterRadarScanMsg registers a callback for radar devices of type RMS1000 and RMS2000. Radar scan messages are not applicable for MRS lidars.

Currently the sick_scan_xd API does not support queries of the contamination status. We will clarify a possible support of ros-like service calls via API with SICK.

janvanz62, 2024-04-03:
thanks for the reply.

So telegrams and the “roslauch” must currently be used.

Q1: Which github project is recommended for a small ARM processor and building the "roslauch" package from scratch?

It takes up to 20 seconds for the MRS1000 to send pointcloud data after starting an application linked to sick_scan_xd.

Q2: Are telegrams only processed after 20 seconds after starting the MRS1000 too?

rostest, 2024-04-03:
Thanks for your reply. We recommend to use ROS-1 or ROS-2 for processing lidar scans and point clouds. See https://ros.org/ for installation, build instructions and support.
Scan data telegrams are processed after receiving them from the lidar, i.e. the first point cloud message will be generated after the initial startup phase is completed.

closed 2024-04-09 by rostest

janvanz62, 2024-05-02:
We tested the azimuth angle correction for the MRS1000 successfully in V3.3.1. 
Two questions regarding contamination status MRS1000: 
Q1 Is the contamination support for the MRS1000 planned for release 3.4.0? 
Q2 When will release with contamination support be expected available?

rostest, 2024-05-02:
Thanks for your feedback. API-Function `SickScanApiSendSOPAS` to query the device state or contamination result is available in the develop branch (default branch) https://github.com/SICKAG/sick_scan_xd/tree/develop.