SICKAG/sick_scan_xd #310: Problems while trying to run the API when compiled as an x86 .dll

Sertzu, 2024-04-03:
Hello,

Due to limitiations elsewhere in our project we are forced to into Win32 (x86).

I am building the .dll without ros support in Visual Studio with a slightly modified command from the Instruction:

```
cd sick_scan_xd-picoscan
set _os=x86
set _cmake_string=Visual Studio 17 2022
set _msvc=Visual Studio 2022
set _cmake_build_dir=buildx86
if not exist %_cmake_build_dir% mkdir %_cmake_build_dir%
pushd %_cmake_build_dir%
cmake -DROS_VERSION=0 -G "%_cmake_string%" -A Win32 ..
cmake --build . --clean-first --config Debug
cmake --build . --clean-first --config Release
REM open sick_scan_xd.sln in Visual Studio 2022 for development and debugging
popd
```

Note that the standard x64 version works without problems and the compilation of the shared library also works. Problems also do not arise when loading the library, however once a library function is called the following exception occurs:

[image]

Furthermore if I choose to ignore this exception and continue anyways the program appears to continue execution in a normal way, however this is not acceptable in our use case since undetectable problems could occur down the line.

Could it be that the library just does not work as an x86 library or am I missing something?

Kind regards
Markus

rostest, 2024-04-03:
Thanks for your feedback. 32-bit architectures are not officially supported. The exception is most likely caused by different calling conventions. If your x86 application uses a different calling convention than the library, a runtime check may fail. See https://stackoverflow.com/questions/6334283/declspec-and-stdcall-vs-declspec-only for details.
Make sure that you build the sick_scan_xd library and your application with identical calling conventions. In Visual Studio it can be set by Project Properties -> Configuration Properties -> C/C++ -> Advanced -> Calling Convention.

Sertzu, 2024-04-09:
Thank you for the answer, however I tested the library directly with the provided test project inside the same solution and made sure that the calling convention is the same. (Which it is by default)

It seems that I will have to push for the project to be upgraded to an x64 architecture.

Also could you maybe place a disclaimer somewhere in the documentation that x86 is not supported so that other folks know in advance?

closed 2024-04-09 by Sertzu

rostest, 2024-04-09:
@Sertzu Thanks for following up. We will add a note about not supporting 32 bit systems in the next release.