SICKAG/sick_scan_xd #315: Tim 7XX ROS2 Integration Question (URDF/Params)

anath93, 2024-04-04:
Hello, 

Environment - ROS2 Galactic 
Architecture - aarch64

Reading through the docs, I have few questions,

How to get complete list of complete configurable parameters ? 

[image]

[image]

Is there .stl file for Tim7xx ? 

Where is the origin/reference of LIDAR for adding tf  ?

Is there any example for Tim7xx urdf ? 

Can this LIDAR work in USB mode too (as per sales executive, they said Yes but I would like to confirm this) ?

anath93, 2024-04-05:
@rostest any input on the above ?

anath93, 2024-04-05:
@rostest Thank you for the detailed reply, what is the origin point on LIDAR so I can do static tf broadcaster for connecting to my base frame ?

anath93, 2024-04-05:
@rostest thank you for the information and manual at the same time. 

Just one last thing before I close the topic, when is the next release due ?

anath93, 2024-04-10:
I am trying to run the lidar on Ubuntu 20.04 ROS2 Galactic side and having some issue while launching the 
"ros2 launch sick_scan_xd sick_tim_7xx.launch.py" 

[image]

[image]

I am able to ping the network 192.168.0.1 and able to get data on SOPAS windows side but not on linux side, Any thoughts ?

rostest, 2024-04-12:
The TiM781 lidar has a constant scan frequency of 15 Hz and an angular resolution of 0.33 degree:

[image]
[image]

Therefore you can not set the scan frequency of the lidar. See https://www.sick.com/de/en/catalog/produkte/lidar-und-radarsensoren/lidar-sensoren/tim/tim781-2174101/p/p594148 and https://cdn.sick.com/media/docs/5/05/905/operating_instructions_tim781_en_im0082905.pdf for details.

The parameters configurable for Tim7xx devices are listed in https://github.com/SICKAG/sick_scan_xd/blob/develop/launch/sick_tim_7xxS.launch.