SICKAG/sick_scan_xd #318: Unicode error

honyu-sick, 2024-04-15:
Dear Support Team,

Currently our customer is facing this unicode error when launching our Picoscan launch file, sincerely seek for your assistance to advise on this enquiry. 

![
[image]

Thanks!

rostest, 2024-04-15:
Thanks for your feedback. The error message "encoding with idna codec failed" means that a DNS query can not be resolved. Most likely the IP address given by parameter `udp_receiver_ip:=192.168.0.100` is incorrect.

Parameter `udp_receiver_ip` must be the ip address of the computer running sick_scan_xd, see https://github.com/SICKAG/sick_scan_xd/blob/develop/doc/sick_scan_segment_xd.md for details. This parameter has to be configured specifically for each system. Use e.g. `ipconfig -a` to find the ip address.

Parameter `udp_receiver_ip:=192.168.0.100` is used in some examples, but is most likely not the correct ip address of your customers computer. If the picoScan lidar has ip address 192.168.1.249, make sure that the lidar and the computer running sick_scan_xd use the same subnet.

Please check the ip addresses of the lidar and the customers computer. Do not hesitate to reply in case the problem remains.

honyu-sick, 2024-04-16:
Dear Support,

Thanks for the prompt response and explanations.

Appreciated =)

closed 2024-05-03 by rostest