SICKAG/sick_scan_xd #319: IMU and Cloud topic in Multiscan100 ROS2 humble

Franklein20, 2024-04-18:
Hello
I try publish the IMU and Cloud topic with the multiscan100
For the IMU , i can see this output from the terminal , i don't know what happens 
` SickScanServices: Request "sWN ScanDataEnable 1" successfully sent, received reply "sWA ScanDataEnable"
[sick_generic_caller-1] [INFO] [1713452065.458930820] [sick_scan_xd]: SickScanServices::sendSopasCmdCheckResponse(): request: "sWN ScanDataEnable 1", response: "sWA ScanDataEnable"
[sick_generic_caller-1] [INFO] [1713452065.458938403] [sick_scan_xd]: SickScanServices: Sending request "sWN ImuDataEnable 1"
[sick_generic_caller-1] [INFO] [1713452065.458949317] [sick_scan_xd]: Sending  : <STX>sWN ImuDataEnable 1<ETX>
[sick_generic_caller-1] [INFO] [1713452065.459963583] [sick_scan_xd]: Receiving: <STX>sFA 3<ETX>
[sick_generic_caller-1] [WARN] [1713452065.460006665] [sick_scan_xd]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX>sWN ImuDataEnable 1<ETX>, received answer: "sFA 3", expected patterns: "sWA ImuDataEnable","sAN ImuDataEnable"
` 
For the cloud topic is empty but i get others topics like `/cloud_all_fields_fullframe
/cloud_polar_unstructured_fullframe
/cloud_polar_unstructured_segments
/cloud_unstructured_fullframe
/cloud_unstructured_segments
`

Could you help me ?

rostest, 2024-04-19:
Thanks for your feedback. SOPAS request `sWN ImuDataEnable 1` activates the IMU telegrams. The lidar replies with error code 3, which means an unknown SOPAS command. Most likely the firmware needs to be updated. See https://www.sick.com/no/en/catalog/products/lidar-and-radar-sensors/lidar-sensors/multiscan100/muls1aa-112211/p/p673512?category=g569793&tab=downloads for downloads.
Alternatively you can deactivate IMU messages in the launchfile by setting `<param name="imu_enable" type="bool" value="False"/>`.

Franklein20, 2024-04-19:
Thanks for your feedback.
Follow the link, i want to update my firmware and i saw the version available is 2.0.0, But in the documention, it mentions this: 
`When upgrading from V1.4.2 and lower to V2.0.0, the function for IMU will be removed. This must be purchased
at a later date.
In some cases, the license for Echo Filter and Contamination Indication are removed, as well. If you want to
update multiple devices, please check if the functions are available after updating one device to V2.0.0. If you
are facing any issues with that, please open a ticket or contact Jisheng Zhang if you would like to activate
functions afterwards.`

I'm afraid if this update remove the IMU fonctionnality. 
Is it correct?

rostest, 2024-04-19:
Thanks for your reply. This information was new to me. Please contact SICK support for an upgrade of your multiScan device.

Franklein20, 2024-04-19:
Ok thanks

closed 2024-04-19 by Franklein20