SICKAG/sick_scan_xd #320: Failed to open TCP connection to 192.168.0.2:2122, aborting. Connect error10060

treeandcat, 2024-04-19:
I currently have two multiScan136 laser radars connected to the same computer. I connected them to different network ports of the computer.
The IP of one of the radars is set to 192.168.0.1, the corresponding network port is set to 192.168.0.222, and the target port is set to 2115
The IP of another radar is set to 192.168.0.2, the corresponding network port is set to 192.168.0.236, and the target port is set to 2116
I modified the corresponding launch file. Because I have seen related issues before, sick_scan_xd does not support opening two radars in one program, so I used two programs similar to minimum_sick_scan_api_client.cpp to drive the radar and obtain point cloud data. Two Except for the configuration information of the radar, other codes in the program are basically the same. I also set the two programs to start automatically at boot.
But one of the radars always displays [Error]: Tcp::open: Failed to open TCP connection to 192.168.0.2:2122, aborting. Connect error10060 The connection attempt failed because the connecting party did not reply properly after a period of time or the connected host was unresponsive.
As a result, this program cannot obtain the point cloud data of lidar.
But in fact, I can ping 192.168.0.2. I can also display point cloud data by entering the IP address on the web page. I can also connect after closing the program and reopening it. However, it always reports this error every time I turn on the computer. The error resulted in the inability to obtain point cloud data, and these two programs must also be started automatically after booting.
May I ask where this problem may occur and how to solve it? I look forward to your reply. Thank you.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

rostest, 2024-04-22:
Thanks for your feedback. Just to make sure we understand the problem correctly: If the application is started twice with different configurations and PC and lidars are both up and running, then lidars and processes behave as expected. However, if the application is started twice immediately after booting, sick_scan_xd cannot establish a TCP connection to the second lidar. Is this correct?

If so, the lidar is probably not yet in operating mode in the second case. Depending on the lidar, the initialisation phase may take a few seconds. The error message from the operating system ("The connection attempt failed because the connecting party did not reply properly after a period of time or the connected host was unresponsive") indicates that the lidar is not (yet) responding to TCP connection requests. In this case, a delayed start of the application or a sick_scan_xd API thread could solve the problem.

treeandcat, 2024-04-22:
We have tried to use std::this_thread::sleep_for(std::chrono::seconds(60)); to delay the startup of any of the sick_scan_xd APIs, but it still prompts such an error.

rostest, 2024-04-23:
Thanks for your reply. The problem and error messages seem to be related to the IP interface during startup. TCP issues during booting are a bit beyond our scope. We recommend to start the sick_scan_xd API delayed and latest in the boot process, e.g. by starting a script via `xterm -e <startscript>` in a .bashrc or at X11-start.

closed 2024-04-23 by rostest

treeandcat, 2024-04-23:
Thank you very much for your reply. We checked the connection between the program and the radar. We found that the two radars are connected to a one-to-two network card, which caused an abnormal tcp connection of one radar when the program started. Currently, the two radars are connected After connecting to the network ports of two different network cards, both radars can be connected normally after being powered on. I also hope it can be a reference for people who encounter the same problem.

rostest, 2024-04-23:
@treeandcat Many thanks for reporting this issue and its solution!