SICKAG/sick_scan_xd #322: LMPoutputRange Problem

CTech-dev0, 2024-04-27:
Hi.

I'm running the driver for linux native (no ROS) and I'm having some issues with the Lidar.

I'm using a Tim561 (V2.51) 
The problem happens when sending 
<STX><STX><STX><STX><Len=0033>sWN LMPoutputRange | 0x00 0x01 | 0x00 0x00 0x0d 0x05 | 0x00 0x00 0xbb 0xa0 | 0x00 0x29 0x32 0xe0 | CRC:<0x9f> 

In the SOPAS ET the device is configured to output data in the 0° - 180° 
The launch file is written with the following parameters
<param name="min_ang" type="double" value="0"\>
<param name="max_ang" type="double" value="3.141592654"\>

But the values in the Telegram do not seem to match the defined parameters. 

And then I receive

<STX>sFA\ x00\x04<ETX> = Sopas_Error_LOCALCONDITIONFAILED

rostest, 2024-04-29:
Thanks for your feedback. There is probably a misunderstanding regarding the coordinate systems used.

sick_scan_xd uses ros coordinates for all lidars for both configuration (min_ang, max_ang) and exported point cloud data. I.e. the x-axis points forward, y to the left, z up. The default parameters in the launchfile sick_tim_5xx.launch are min_ang=-2.35619449 (-135 deg) and max_ang=2.35619449 (+135 deg), i.e. valid angles are within the range from -135 deg up to +135 deg. The parameter max_ang=3.141592654 resp. 180 deg is therefore invalid.

TiM5xx lidars use internal coordinates with an offset of 90 degrees to ros coordinates. Valid angles are between -45 deg up to +225 deg in lidar coordinates:
[image]

Use min_ang=-1.178097245 (-90 deg) and max_ang=1.178097245 (+90 deg) in the launch file in ros coordinates. This corresponds to an angle range of 0 to 180 degree in lidar coordinates used by "sWN LMPoutputRange" (with a 90 degree offset between ros and lidar coordinates).

CTech-dev0, 2024-05-01:
Thanks !
Now I'm able to run the generic caller. 

I'm having other issues with the library. Is okay to post them here instead of making a new issue ?

Regards.

rostest, 2024-05-01:
Thanks for your feedback. Problems are often correlated, keep going.

closed 2024-05-27 by rostest