SICKAG/sick_scan_xd #326: How to use PicoScan ROS1 to run 2 PicoScan for negivation?

sjeffreydong, 2024-05-09:
Dear Teams,
Would you please share with us the workflow how to use 2 PicoScan for negivation ?
Customer used our 2 PicoScan can't work well, they can't start together.

[image]

rostest, 2024-05-10:
Thanks for your feedback. To use 2 picoScan lidars under ROS-1, start sick_scan_xd twice with different ip addresses and topics in two ros nodes. See the [FAQ](https://github.com/SICKAG/sick_scan_xd/blob/develop/FAQ.md#how-to-run-multiple-sensors-concurrently) for an example configuration.

sick_scan_xd publishes PointCloud2 and LaserScan messages, SLAM and navigation using two lidars is a bit beyond the scope of the driver. See [slam.md](https://github.com/SICKAG/sick_scan_xd/blob/develop/doc/slam.md) and [#206](https://github.com/SICKAG/sick_scan_xd/issues/206) for more information or use [run_linux_ros1_simu_slam_mrs1xxx.bash](https://github.com/SICKAG/sick_scan_xd/blob/develop/test/scripts/run_linux_ros1_simu_slam_mrs1xxx.bash) resp. [test_200_slam_ros1_hector.launch](https://github.com/SICKAG/sick_scan_xd/blob/develop/launch/test_200_slam_ros1_hector.launch) as an example.

sjeffreydong, 2024-05-29:
Dear Teams,

Our customer is running 2 picoScan150 on one controller, however will happen issues with one can't start and no data.
[image]

rostest, 2024-05-29:
Thanks for your feedback. The photo does not support a clear diagnosis of the problem. Please send the launchfiles used for both picoScan150 devices, the commandline calls and parameters to launch sick_scan_xd and complete logfiles for further diagnosis.

sjeffreydong, 2024-05-29:
Dear Teams,

Please have a look. Could you check what is the issues ? why 2 picScan can’t run well on the one controller when start 2 pcioScan together ? 
[picoScan150.zip](https://github.com/SICKAG/sick_scan_xd/files/15480153/picoScan150.zip)
mail

sjeffreydong, 2024-05-29:
> EXTERNAL EMAIL: This message was sent from outside of SICK. Please do not click any links or open any attachments unless you trust the sender and know the content is safe. Please report suspicious emails via SMAX<https://userservice.sick.com/saw/ess/offeringPage/160523?query=phishing>.

Dear Teams,

Could you check what is the issues ? why 2 picScan can’t run well on the one controller when start 2 pcioScan together ?

Best regards

Jeffrey Dong
Product Manager | Sales & Service China & Hong Kong
SICK China Co., Ltd.  |  Industrial Park   |  Building A, Suzhou Center Plaza  |  SuZhou, 215021  |  China
广东西克智能科技有限公司苏州办事处  |  江苏省苏州市工业园区苏州中心广场A幢1005室， 215021
Phone +86 512 67339115-5335 | Mobile: +86 186 1621 9594  |  ***@***.***  |  Service hotline 4000 121 000  |  www.sickcn.com  |  https://www.sick.com/cn/zh/cart

rostest, 2024-05-29:
Thanks for further information and launchfiles. To run two picoScan devices simultanously, sick_scan_xd must be started in two different nodes. Each node must be configured with different lidar ip addresses and udp ports, different node names, different ros topics and frame ids for each point cloud. Therefore the following launchfile parameter should be overwritten by individual settings for each lidar:
* "hostname": e.g. "192.168.0.190" and "192.168.0.98"
* "nodename": e.g. sick_picoscan0" and "sick_picoscan1"
* "publish_frame_id": e.g. "world0" and "world1"
* "publish_laserscan_segment_topic": e.g. "scan0_segment" and "scan1_segment"
* "publish_laserscan_fullframe_topic": e.g. "scan0_fullframe" and "scan1_fullframe"
* "imu_topic": e.g. "imu0" and "imu1"
* "udp_port": e.g. "56661" and "56662"
* "imu_udp_port": e.g. "7503" and "7504"
* individual topics and frame id for each customized point cloud, e.g.
    * replace all "topic=/cloud_" by "topic=/cloud0_" resp. "topic=/cloud1_"
    * replace all "frameid=world" by "frameid=world0" resp. "frameid=world1"

Two modified launchfiles are attached. We recommend to first verify the launchfile configuration separately for each picoScan, then run both devices simultanously.
[picoscan_launchfiles.zip](https://github.com/SICKAG/sick_scan_xd/files/15484846/picoscan_launchfiles.zip)

Parameter udp_receiver_ip must be the ip address of your PC running sick_scan_xd. We recommend to use the same subnet as the lidar, i.e. to change the customers current ip address 224.0.0.1 to an ip address within the subnet 192.168.x.y.

closed 2024-07-08 by rostest