SICKAG/sick_scan_xd #328: could not complete the  picoScan launch process without any error report.

Jonny0816, 2024-05-15:
Dear sir
When I launch picoScan,it always stop in the process but without error report.
Ping the picoScan is ok and I tested the driver with TiM240 is ok.Seems the Ethernet connection and build driver are ok.
please refer to below file of the launch streaming and help to analysis what is the reason.
[picoScan launch streaming record.md](https://github.com/SICKAG/sick_scan_xd/files/15318591/picoScan.launch.streaming.record.md)

rostest, 2024-05-15:
Thanks for your feedback and log file. The picoScan lidar was successfully initialized via TCP, but did not send any scan data via UDP. Most likely the given parameter `udp_receiver_ip` is incorrect. 

Parameter `udp_receiver_ip` must be set to the ip address of the local machine, i.e. the computer running sick_scan_xd.  This is the IP address to which the lidar sends the scan data via UDP. You can find it e.g. with `ifconfig -a`. 

The call `roslaunch sick_scan_xd sick_picoscan.launch hostname:=192.168.0.1 udp_receiver_ip:=192.168.0.100` is often used as an example. Replace `192.168.0.100` with the ip address of your local machine running sick_scan_xd.

Jonny0816, 2024-05-21:
Thanks for the recommend and that is the reason, Before I used the PC IP and solved this question after switch to Vmware IP.

closed 2024-05-21 by rostest