SICKAG/sick_scan_xd #329: Failed to build on windows ROS2

kjoelovelife, 2024-05-16:
Hi there,

I encountered an issue while trying to install sick_scan_xd on Windows 10 with ROS2 Humble.

The error message I received is as follows:

[image]

I have ensured that the path C:/sick_scan_ws/src/sick_scan_xd/include exists.

rostest, 2024-05-18:
Thanks for your feedback. The path `C:/sick_scan_ws/src/sick_scan_xd/include_directories` looks strange as `include_directories` is not a subfolder in sick_scan_xd. Maybe it is from a previous build?

Please remove all temporary build folders, i.e. run
```
cd C:\sick_scan_ws
rmdir /s/q .\build
rmdir /s/q .\install
rmdir /s/q .\log
del /f/q .\src\CMakeLists.txt
```
and rebuild as described in [Build on Windows ROS2](https://github.com/SICKAG/sick_scan_xd/blob/develop/INSTALL-ROS2.md#build-on-windows-ros2):
```
call "%ProgramFiles(x86)%\Microsoft Visual Studio\2019\Community\Common7\Tools\VsDevCmd.bat" -arch=amd64 -host_arch=amd64
call C:\opt\ros\humble\x64\setup.bat
colcon build --packages-select sick_scan_xd --cmake-args " -DROS_VERSION=2" --event-handlers console_direct+
call .\install\setup.bat
```
Please post a logfile of the entire build process if the problem persists.

rostest, 2024-05-22:
Many thanks for following up. We have adapted CMakeLists.txt for ROS-2 humble on Windows. Please use the update in branch [feature/windows_ros2_humble](https://github.com/SICKAG/sick_scan_xd/tree/feature/windows_ros2_humble) and rebuild.

closed 2024-07-08 by rostest