SICKAG/sick_scan_xd #334: MultiScan100: Could not get initial transform from base to laser frame, "world_6_2" passed to lookupTransform argument source_frame does not exist.

mmaeg, 2024-05-21:
Hello,

I am contacting you, because I am having troubles with my multiscan100 lidar sensor. Indeed, I would like to use it to do SLAM navigation. I have already develop a SLAM algorithm on ROS that work with several lidar sensors, such as sick NAV245. However, when I try to run the SLAM program with the multiscan, a Warning message appears, and tells me that the source_frame "world" does not exist (even if this is the default source frame in the launch file). Because of this warning, the data from the lidar are not communicated to the SLAM program.

From the analysis of the warning message, and the driver source code, I think the problem is related to the tf package of ros. Indeed, the multiscan is publishing a tf topic, when other lidar sensors I have used did not. Moreover, the topic of the lidar's data is communicated correctly.

I join to this message few screenshot of my situation.

if you have any idea of why do I have this problem, and how to solve it, could you please help me.
[image]
[image]
[image]

rostest, 2024-05-21:
Thanks for your feedback. The multiScan100 lidar scans 16 layers, each with a different elevation angle. Since ROS LaserScan messages only support one layer, sick_scan_xd can publish different LaserScan messages with different frame ids for each layer. The default frame ids are ""world_<layer_idx>" (use `rostopic echo /multiScan/scan_fullframe | grep -i frame_id` to display the Laserscan topics).

Using the default configuration sick_multiscan.launch, sick_scan_xd publishes Laserscan messages of the high resolution layer index 6 on topic "/multiScan/scan_fullframe" with frame id "world_6". Use a ROS static_transform_publisher to define the transform between the multiscan frame id "world_6" and your base frame.

rostest, 2024-05-21:
Thanks for your reply and information. If echos are enabled, sick_scan_xd will publish a Laserscan message for each echo with frame ids `world_<layer_idx>_<echo_idx>`. If echos are not needed, it is recommended to activate an echo filter with `<param name="host_FREchoFilter" type="int" value="2" />` in the launchfile. This will enable the last echo and is the default. Make sure you use the latest version of sick_scan_xd. Using the default echo filter, the frame id of the Laserscan message is "world_6".

The transform chain for SLAM is system specific and not part of sick_scan_xd. The file [test_200_slam_ros1_hector.launch](https://github.com/SICKAG/sick_scan_xd/blob/develop/launch/test_200_slam_ros1_hector.launch) shows an example configuration for the hector SLAM algorithm implemented in ROS1. The static_transform_publishers in this file define the transform chain and the frame ids. The parameters of these static_transform_publishers must be adapted to the robot setup and the lidar frame ids. See https://github.com/SICKAG/sick_scan_xd/blob/develop/doc/slam.md for further information.

Note that SLAM works independently from sick_scan_xd; sick_scan_xd just publishes the point clouds and scan data. Therefore, you cannot modify the source code in the sick_scan_xd folder to provide the transform chain required by SLAM. Use the SLAM configuration to adapt the transforms to your robot setup and multiScan lidar frames.

mmaeg, 2024-05-22:
Hello,

I have a question concerning the content of the topic /scan_fullframe (I renamed it /scan) published by the node of the multiscan. Indeed, when using the Compact data format, and looking the content of the topic, the time stamp of every publishment is equal to 0. The following screenshot show the content of the topic:
[image]
I have not modified any thing in the source code of the multiscan node, only the launch file. Is it normal, or do I have to check or change something to have the real time stamp being published ?
Indeed, when doing some trials for the SLAM, I have seen that it could be a problem for transformations between frames. Because the /tf topic is collecting different transformations between different frames, it requires the time stamp in order to synchronize the different nodes doing the transformations.
Thank you in advance for your response.

mmaeg, 2024-05-22:
Hello,
I have the latest version of sick_scan_xd. I finally solve the problem, I manage to replace msg_data.timestamp_sec by a ros::time::now. This change makes the SLAM program working well.  For the layer, as I need to have a world frame, I change in the same way the declaration of the frame by removing its layer number. 
Thank you for your help, I close this issue.

closed 2024-05-22 by mmaeg