SICKAG/sick_scan_xd #335: How activate the IMU frame on Lidar MRS1000

Franklein20, 2024-05-23:
I am working on integrating the MRS1000 LiDAR with ROS2 Humble, and in addition to point clouds, I also want to utilize the IMU. With the correct configuration, I have the topics for both point clouds and IMU data.

My question concerns the frames for these two types of data. The point clouds are published on the 'cloud' frame, and the IMU data is on the 'imu_link' frame (as indicated in the topics). How can I activate or create the frame for the IMU, because when I visualize the frames, there is nothing displayed for 'imu_link'?

rostest, 2024-05-24:
Thanks for your feedback. Point cloud messages of type sensor_msgs/PointCloud2 are published on topic "/cloud" with frame id "cloud". IMU messages of type sensor_msgs/Imu are published on topic "/sick_mrs_1xxx/imu" with frame id "imu_link". You can view IMU messages with `rostopic echo -p /sick_mrs_1xxx/imu` or visualize them with rviz by adding topic "/sick_mrs_1xxx/imu" and setting option "Fixed Frame" to "imu_link". Use a static_transform_publisher like `rosrun tf static_transform_publisher 0 0 0 0 0 0 base_frame imu_link 100` to link imu frames to another base frame.

[image]

Franklein20, 2024-05-31:
ok i see.
Thanks

closed 2024-07-08 by rostest