SICKAG/sick_scan_xd #336: (Picoscan) error message showing data missing on echo 0

honyu-sick, 2024-05-27:
Dear SICK Team,

Kindly need your assistance to advise on our customer's current issue. 

Set the correct output port for the measurement data on the utilities of the front sensor. 

The utilities can show the scanning data. 
[image]

Set the corresponding parameters in the config file and the launch file. The launch file is attached in the attachment. 
 

       3. launch the launch file, got the error message showing data missing on echo 0, which is the first echoing action. The /sick_multiscan/scan topic for the scan data is also not publishing 

 
[image]
           

 

      4. Try to echo the imu data, able to receive the imu data, meaning the connection is established there. 
[image]

Many Thanks!

Regards
Hon

rostest, 2024-05-27:
Thanks for your feedback. The error message is generated by a msgpack validator, which can optionally be used to test msgpack data. This option is not supported for compact data and therefore disabled by default. Please use the default settings in the launch file for compact format, i.e.:
```
<arg name="scandataformat" default="2"/>
<param name="msgpack_validator_enabled" type="bool" value="False" />
```

Laserscan messages are published in the topics `/sick_picoscan/scan_fullframe` (picoScan full frame scans) and `/sick_picoscan/scan_segment ` (picoScan segment scans).

honyu-sick, 2024-05-28:
Dear Support Team,

Thankful for your support 

As follow the advice given, the customer have edited the new launch file with the MsgPack Validator disabled. In the attachment is the new launch file. 

The error message is gone. But the topic /sick_picoscan/scan_segment and /sick_picoscan/scan_fullframe still have not output publishing. The imu topic is ok. 

They also log the start up logs in the beginning of the startup, maybe this files will help to advice further.

 
[launch file (new param).txt](https://github.com/SICKAG/sick_scan_xd/files/15463570/launch.file.new.param.txt)
[lidar startup logs.txt](https://github.com/SICKAG/sick_scan_xd/files/15463571/lidar.startup.logs.txt)

honyu-sick, 2024-05-29:
After using the picoScan.launch file in the package and updated to 3.4.0, the lidar scan is ok and both the readings can be read by ROS. 

However, under the picoscan, there are 2 topics, for the data by "fullframe" mode and "segment" mode, what is the difference between the 2 mode?  

From launch file,sick_picoscan.launch ,  stated fullframe is the data scans and segment refer to the degree of the scans, is that correct?

If I want to use the scan data to create the navigation map for the navigation software, which is the mode "fullframe" mode or segement I need to use?

rostest, 2024-05-29:
Thanks for further information. picoScan provides segmented scan data, where each segment covers a part of the full scan. The "segment" topics contain the scan points of one segment. The "fullframe" topics cumulate all segments into one point cloud, which covers the full azimuth range from -138.0 to +138.0 degrees. The "fullframe scan" in topic "scan_fullframe" is recommended for navigation.

closed 2024-07-08 by rostest