SICKAG/sick_scan_xd #339: How to narrow the range of azimuth angles

kazuse5, 2024-05-31:
Hello,

We are evaluating picoScan with ROS2, but does not seem to be able to change the "Scan Range Filter".
Would you please tell us how to narrow the range of azimuth angles ?

[image]

 ROS2 humble
 picoScan V1.3.0 (PN1134609)
 sick_scan_xd 3.4.0

The following site say.
"An azimuth filter is currently not supported for picoScan lidars."
https://github.com/SICKAG/sick_scan_xd/issues/266

When is it expected to improve ?
Thank you in advance.

rostest, 2024-05-31:
Thanks for your feedback. As far as we know, an azimuth filter (LFPangleRangeFilter) is not supported by picoScan lidars. We will clarify with SICK whether support for this feature is planned.

gschwsiSICKAG, 2024-05-31:
As an alternative option, the picoScan150 offers native ROS2 (DDS) to connect to ROS2 Humble without the need for an external driver. With this feature, you can use the scan range filter. Please use https://supportportal.sick.com/ to contact SICK support.

closed 2024-07-08 by rostest