SICKAG/sick_scan_xd #342: Is there a way to combine all echos in picoscan150 in a single world frame

logesh-gtx, 2024-06-05:
When i Switch to all echos mode the the `scan_fullframe` topic data comes with three different frame id's, but when i change to first echo / last echo it works fine but with only limited to one echos data.

**`For Single echo (first / last echo) topic data`**
param: `<param name="host_FREchoFilter" type="int" value="2" /> `
[image]

**`For ALL echos filter topic data`**
param: `<param name="host_FREchoFilter" type="int" value="1" /> `
[image]

rostest, 2024-06-05:
Thanks for your feedback. Topic scan_fullframe publishes the scan data using LaserScan messages. LaserScan messages are defined by the ROS standard in https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/LaserScan.html. Due to their format, these messages are limited to one planar scan with one layer and one echo. Multiple layers and echos are therefore published by multiple LaserScan messages with different frame ids. The frame_id of a LaserScan message is `<base_frame_id>_<layer_id>` for a single echo configuration (first or last echo) and  `<base_frame_id>_<layer_id>_<echo_idx>` for multiple echos.

If `<base_frame_id>` is "lidar_1", the frame_id is "lidar_1_1" for the single echo configuration (first or last echo) or "lidar_1_1_0" (first echo), "lidar_1_1_1" (2.nd echo), "lidar_1_1_2" (last echo) for multiple echos (host_FREchoFilter=2).

If your application does not require multiple echos, we recommend to use the last echo (default parameter host_FREchoFilter=1).

To process multiple echos in one scan, we recommend to use the point cloud messages (data type PointCloud2) published in topic cloud_unstructured_fullframe (fields x,y,z,i) or cloud_all_fields_fullframe (all fields incl. x,y,z,i,range,azimuth,elevation,layer,echo).

logesh-gtx, 2024-06-14:
Thank you for your response, it works good in `/cloud_unstructured_fullframe`

closed 2024-07-08 by rostest