SICKAG/sick_scan_xd #345: LMS511 ROS1 run problem

nicolas-simon, 2024-06-11:
Hi everyone !

Like the issue #298, I've got the same  error code : 

[ INFO] [1677781569.128073088]: SickScanServices: ros services initialized
[ INFO] [1677781569.128097472]: Setup completed, sick_scan_xd is up and running. Pointcloud is published on topic "cloud"
[ INFO] [1677781569.795537856]: Software PLL locking started, mapping ticks to system time.
[ INFO] [1677781569.795580800]: 1 / 6 packets dropped. Software PLL not yet locked.
[ERROR] [1677781569.795719616]: ## ERROR SickScanMessages::parseLFErecMsg(): parse error, fields number = 0 should be greater than 0 (/home/nvidia/dock_ws/src/sick_scan_xd/driver/src/sick_scan_messages.cpp:267)
[ERROR] [1677781569.795749984]: ## ERROR SickScanMessages::parseLFErecMsg(): 22 byte received: \x02\x02\x02\x02\x00\x00\x00\x0dsSN LFErec \x00\x00U
[ WARN] [1677781569.795775712]: ## ERROR SickScanCommon: parseLFErecMsg failed, received 22 byte LFErec \x02\x02\x02\x02\x00\x00\x00\x0dsSN LFErec \x00\x00U
[ INFO] [1677781569.808529952]: 2 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1677781569.825219552]: 3 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1677781569.844034240]: 4 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1677781569.864875424]: 5 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1677781569.884235680]: Software PLL is ready and locked now!

Unfortunately, I need the field monitoring. 

The LMS511 firmware is up to date (v 2.3).

Also, I did get any messages on topics, even if I put False at activate_lferec param : 
/sick_lms_5xx/cloud
/sick_lms_5xx/lferec
/sick_lms_5xx/lidoutputstate
/sick_lms_5xx/marker

But I've got message on topics :
/cloud
/scan

Anyone have an idea of this issue ?

Kind regards,
Nicolas

rostest, 2024-06-12:
Thanks for your feedback. Please make sure the evaluation fields have been set up with SOPAS ET. Are the fields shown correctly in SOPAS ET? A complete logfile incl. all field initialization messages after driver start is helpful for further analyzation. Please check the field setup in SOPAS ET and post a complete logfile.

nicolas-simon, 2024-06-12:
Hi,

Thanks for your quick reply !

Effectively, the evaluation fields has an issue in the SOPAS ET. Now I've resolved this issue, I can get the message of the evaluation field from the topics lferec.

Thanks for your help !

Nicolas

closed 2024-06-12 by nicolas-simon