SICKAG/sick_scan_xd #346: Issue building the shared generix library on Ubuntu 24.04

jashangills, 2024-06-12:
I am encountering this  error while attempting to build the shared generic library on Ubuntu 24.04. 

It seems that the compiler is unable to recognize the uint8_t type, which is defined in the <cstdint> header file. The error suggests that the #include <cstdint> directive may be missing from the dataDumper.cpp, dataDumper.h files.

[image]

I mangaed to do a successful Build for libsick_ldmrs  library but cant do the same for generic one

Any help/suggestions will be appreciated.

rostest, 2024-06-12:
Thank you very much for this error report. Can you include the standard library header `<cstdint>` in file dataDumper.h for a quick check, i.e. add a line `#include <cstdint>` at the beginning?

jashangills, 2024-06-12:
I did try adding it this morning, and It didnt resolve the issue. I have tried it again now to capture the error log. Please see below
[image]

Just wondering,Could this be a gcc issue? ( gcc version 13.2.0 (Ubuntu 13.2.0-23ubuntu4) )

rostest, 2024-06-12:
Thanks for following up. We will investigate the error in more detail.

jashangills, 2024-06-12:
I believe it was the gcc version issue as I successfully built the library on Linux Ubuntu 24.04 by downgrading the GCC compiler to version 12.3.1. After reviewing the release notes for GCC 13.2.0, I discovered that this version requires explicit definitions for headers such as `<cstdint>  and that may have been causing issues while building (https://gcc.gnu.org/gcc-13/porting_to.html)

Downgrading to GCC 12.3.1 resolved this issue and allowed the build to complete successfully.

closed 2024-07-08 by rostest