SICKAG/sick_scan_xd #348: Fail to open TCP connection

BarryLoveBerry, 2024-06-17:
Hi all,
I met a issue with TCP connection failed to open. Here is my PC IP,Lidar IP and  lanunch file hostname IP.
[image]
[image]
[image]

I used the command as below to test.
cd ./sick_scan_ws
export LD_LIBRARY_PATH=.:`pwd`/build:$LD_LIBRARY_PATH  # append absolute path to the build folder
./build/sick_scan_xd_api_test ./sick_scan_xd/launch/sick_lms_4xxx.launch hostname:=192.168.8.35

BarryLoveBerry, 2024-06-17:
Here is the error screenshot.
[image]

rostest, 2024-06-17:
Thanks for your feedback. The error message means that sick_scan_xd cannot establish a TCP connection to the lidar using the IP address 192.168.8.35 on port 2112. Please ensure that the lidar IP address and TCP port are correct and accessible from your PC. Use SOPAS ET to check or modify the lidar network settings. Can SOPAS ET connect to the lidar and display a point cloud?

BarryLoveBerry, 2024-06-18:
> Thanks for your feedback. The error message means that sick_scan_xd cannot establish a TCP connection to the lidar using the IP address 192.168.8.35 on port 2112. Please ensure that the lidar IP [...]

Thanks for replying. I resolved this issue. In addition, I want to transform the lidar data into PCL type. Is there has any instruction files? If you have, could you share it with me? Thanks in advance.

rostest, 2024-06-18:
Thanks for your reply. Sick_scan_xd does not provide a conversion to PCL types. You can use the point cloud handling in https://github.com/SICKAG/sick_scan_xd/blob/develop/test/src/sick_scan_xd_api/sick_scan_xd_api_test.cpp as an example. The structure of the data type SickScanPointCloudMsg is quite similar to the ROS point cloud of type sensor_msgs::PointCloud2. Project http://wiki.ros.org/pcl_ros can be helpful.

UdayVerma20, 2024-06-23:
> Thanks for replying. I resolved this issue. In addition, I want to transform the lidar data into PCL type. Is there has any instruction files? If you have, could you share it with me? Thanks in [...]

How did you fix the issue?

BarryLoveBerry, 2024-06-24:
> How did you fix the issue?

export env LD_LIBRARY_PATH

BarryLoveBerry, 2024-06-24:
Now I created a tool to get the data from lidar LMS4000. and I already received the data from lidar. How to analyze the data and how to transform it into pointcloud(PCL)? Do you have any docs for this issue?
I noticed the my data that I received is a bit different with the official doc. Can you help me to locate the where is not correct? thks.[
[Uploading baowen.txt…]()
](url)

BarryLoveBerry, 2024-06-24:
Now I created a tool to get the data from lidar LMS4000. and I already received the data from lidar. How to analyze the data and how to transform it into pointcloud(PCL)? Do you have any docs for this issue?
I noticed the my data that I received is a bit different with the official doc. Can you help me to locate the where is not correct? thks
[baowen.txt](https://github.com/user-attachments/files/15950821/baowen.txt)

rostest, 2024-06-24:
Thanks for further information. File [baowen.txt](https://github.com/user-attachments/files/15950821/baowen.txt) shows the lidar LMDscandata telegram. You can use the [telegram listing](https://cdn.sick.com/media/docs/7/27/927/telegram_listing_telegram_listing_ranging_sensors_lms1xx_lms5xx_tim2xx_tim5xx_tim7xx_lms1000_mrs1000_mrs6000_nav310_ld_oem15xx_ld_lrs36xx_lms4000_lrs4000_multiscan100_picoscan100_en_im0045927.pdf)   to analyze these data and convert them into the PCL format. We recommend to use the sick_scan_xd API and to convert the scan data received by a registered callback. See [sick_scan_api.md](https://github.com/SICKAG/sick_scan_xd/blob/develop/doc/sick_scan_api/sick_scan_api.md) and the C++ example in [sick_scan_xd_api_test.cpp](https://github.com/SICKAG/sick_scan_xd/blob/develop/test/src/sick_scan_xd_api/sick_scan_xd_api_test.cpp) for details.

BarryLoveBerry, 2024-06-26:
> Thanks for further information. File [baowen.txt](https://github.com/user-attachments/files/15950821/baowen.txt) shows the lidar LMDscandata telegram. You can use the [telegram [...]

Now I got the telegram distance and angle data. How to transform this format to xyz format?

rostest, 2024-06-26:
You can convert polar to cartesian coordinates using the following formulas:
```
x = distance * cos(elevation) * cos(azimuth)
y = distance * cos(elevation) * sin(azimuth)
z = distance * sin(elevation)
```

[sick_scan_api_converter.py](https://github.com/SICKAG/sick_scan_xd/blob/e69f94fea821f4fbe6a6e8eed05c31b4d2ea1bc9/test/python/sick_scan_xd_api/sick_scan_api_converter.py#L126) shows an example.

The elevation angle is 0.0 for 2D lidars like LMS4000. In this case the cartesian coordinates can be converted by
```
x = distance * cos(azimuth)
y = distance * sin(azimuth)
z = 0
```

closed 2024-07-08 by rostest