SICKAG/sick_scan_xd #356: 2 laser in C++ don't work

maurizioscolasipi, 2024-06-28:
Good afternoon,
   I'm trying using 2 laser (LMS500 and LMS4000) in the same process.
I see that the first works but the second doesn't work correctly (the init is ok but the callback doesn't receive any data).
Could you help me?
Thank you
Maurizio
[cpp.zip](https://github.com/user-attachments/files/16034016/cpp.zip)

rostest, 2024-07-01:
Thanks for your feedback. Running multiple lidars in one process via sick_scan_xd API is not supported, see https://github.com/SICKAG/sick_scan_xd/blob/develop/doc/sick_scan_api/sick_scan_api.md.

[image]

We recommend to use ROS, or to run 2 lidars in 2 separate processes via API and interprocess communication.

maurizioscolasipi, 2024-07-01:
Good morning,
   I think the builder of a sensor has to provide the complete driver to allow the user to implement the solution.
I think ROS is a good solution only if you use many 3D sensors, because the overload using ROS is important.
Many years ago I used 2 LMI laser in the same solution, and it work without problem.
Best Regards

[Immagine che contiene testo, Carattere, schermata, logo  Descrizione generata automaticamente]

Maurizio Scola

Vision and automation engineer

Phone     +39 0331 791275

Fax        +39 0331 786050

Web        www.sipi.it<http://www.sipi.it/>

Mail         ***@***.******@***.***>

image.png (view on web)<https://github.com/SICKAG/sick_scan_xd/assets/46743496/568c24eb-34aa-42f9-adfb-797bc518cf86>

rostest, 2024-07-01:
Thanks for your reply. Please contact SICK support for feature request.

closed 2024-08-15 by rostest