SICKAG/sick_scan_xd #359: How to use the ROS topic to reboot Tim561/Tim781S

jimu-baoping, 2024-07-09:
1. Occasionally occurs after the Tim561/Tim781S power-on, it will keep flashing red light, and the master controller can not got point cloud date. How can I solve this problem?
2. Try to use the super account in SOPAS software, reboot the radar, the red light will no longer flash, the master controller can get the point cloud data.
How to use the ROS topic to reboot Tim561/Tim781S ?  The master controller use ubuntu and ROS system.

rostest, 2024-07-10:
Thank you for your feedback. If the red light continues to flash, restart the lidar (power off and on). If the problem persists, reset the lidar to factory defaults. If the lidar is installed in a case, check the temperature for possible overheating. In rare cases, an insufficient power supply can cause problems.

Use SOPAS ET 
* to check if a firmware update is available (and update if so), and
* for diagnostic messages.

If the problem persists, please contact SICK support.

A reboot can in principle be triggered by a SOPAS command via the ROS service, see https://github.com/SICKAG/sick_scan_xd/blob/develop/USAGE.md#ros-services for details. However, this requires both an initialised and running lidar and driver, so it probably won't help if the red light is flashing.

jimu-baoping, 2024-07-11:
Thank you for response.
If the red light continues to flash, restart the lidar (power off and on). The problem will be solved.
The lidar is mounted on the surface of the equipment and there is no problem of heat dissipation.
We have tried to provide more power to the lidar , but the problem remains.
The lidar firmware has been updated,but the problem remains.
Can be configured as if the red light is flashing for a while, lidar will automatically reinitialize? (Automatic reboot?)
Best regards!

rostest, 2024-07-11:
Thanks for your reply. The sick_scan_xd driver monitors the lidar telegrams and automatically restarts and reinitialises the lidar after a timeout (150 seconds without telegrams from the lidar by default). However, a successful software reset requires the lidar to be in operational mode. If the red light continues to blink until rebooting (power off and on), this indicates a hardware problem. Please contact SICK support.

closed 2024-08-15 by rostest