SICKAG/sick_scan_xd #361: Scanner timeout in standby mode

jashangills, 2024-07-15:
Hi, just need some advice regarding the following

When the LiDAR is in standby mode, either after the stopMeas command has been called (I am using SickScanApi with python 3.12) or before the point cloud messages have been published, the TCP receive thread finishes after an inconsistent amount of time I get the following error

[image]

Is there a way to override or change the timeout period that causes the TCP receive thread to trigger and deinitialize the scanner? If so, how can this be achieved?
Also is there a certain time after which this is expected?

Thanks in advance

rostest, 2024-07-16:
Thanks for your feedback. The sick_scan_xd driver monitors the lidar telegrams and restarts and reinitializes after a timeout of 150 seconds by default. You can disable the monitoring with message_monitoring_enabled=False and/or setting the timeout to the max. value 2147483647 in the launchfile:
```
<param name="message_monitoring_enabled" type="bool" value="False" />     <!-- Enable message monitoring with reconnect+reinit in case of timeouts, default: true -->
<param name="read_timeout_millisec_kill_node" type="int" value="2147483647"/> <!-- 150 sec pointcloud timeout, ros node will be killed if no point cloud published within the last 150 sec., default: 150000 milliseconds -->
```
See https://github.com/SICKAG/sick_scan_xd/blob/develop/USAGE.md#driver-states-timeouts for details.

jashangills, 2024-07-22:
Thank you for your response. We’ve encountered the same error both during startup and intermittently during measurements. Could this issue be related to a network problem?

rostest, 2024-07-22:
Thank you for your reply. Network problems can indeed cause timeout errors.

Wireshark is a powerful network diagnostic tool. Start Wireshark, select the ethernet interface used for the lidar connection and watch the network traffic between sick_scan_xd and your lidar. As long as the lidar is sending data via tcp (or udp in case of multiScan or picoScan), sick_scan_xd should not timeout. If Wireshark does not see any tcp data from the lidar, the most likely cause is a network error. 

If you encounter sick_scan_xd timeout errors while your lidar is still sending data, please let us know. In this case, please save and send a complete sick_scan_xd logfile and the network traffic captured by wireshark as a pcapng file. 

Which lidar are you using? If you are using an older version of the sick_scan_xd driver, we recommend that you update to the latest version 3.5.

jashangills, 2024-07-24:
Thank you for your reply. We are in the middle of testing and if the problem persists I will let you know
We are using LMS41xxx

spark-res, 2024-08-15:
Hi @rostest, I'm working with @jashangills on this topic. We're still seeing these intermittent TCP issues and trying to narrow down the cause.

In our deployment environment we are encountering this issue intermittently during an active measurement (possibly standby too).

[image]

In this instance, no stop command has been issued by the software - the TcpRecvThread finishing is unexpected.

Things we have tried:

* Swapping Lidar power source.
* Reducing callback rate (mean filter >1). All other indicators suggest software is keeping up.
* Disabling the message monitoring via the launch file.
* Running the same code (same driver/python versions) on another PC (separate network). All ok.
* Capturing traffic using Wireshark:
    * Shows Lidar TCP messages do stop.
        * [image]

    * Exit seems graceful (FIN, ACK) messages?
        * [image]

    * Raw pcapng file. Note: lidar should be scanning during the entirety of this recording, but stops mid way through due to issue above.
        * Lidar IP is at .102
        * [troubleshooting-lidar-timeout-12-no-stop.pcapng](https://resonatesystems.sharepoint.com/:u:/s/ResonateSystemsExternal/EfJo97bTkG5MpHuxx9HnKEEBAuID1Lg4ajF4YuxAPUrIxg?e=bX3gCN)

Currently our next step is to bypass the network/switch and go straight Lidar to PC. Is there any other troubleshooting steps you might be able to suggest?

Thanks!

rostest, 2024-08-15:
Thanks for following up and further information! Your investigation and procedure look very solid and thorough. 

The pcapng file shows the last scandata from the lidar at time 27.1394 followed by a [FIN] message from the PC (i.e. sick_scan_xd) to the lidar at 27.1399:

[image]

This does not look like an exit due to timeouts. Can you see some error message or any other reason that might have caused this exit? Did sick_scan_xd stop and exit or re-initialize the TCP socket? Can you post the full logfile (i.e. the console output your screenshot shows including previous messages), if it is still available?

spark-res, 2024-08-16:
Thanks @rostest,

Here is a full screen capture of the console output, but unfortunately no text log. (Interleaved other app messages).

[image]

It seems to be only the _TcpRecvThread_ message. No additional exit or re-initialise messages.

After the _TcpRecvThread_ message the app still runs, but no more pointclouds are generated.

rostest, 2024-08-16:
Thanks for your reply. The output in the screenshot is unexpected; there should be some error or informational messages before sick_generic_laser and TcpRecvThread exit.

To reduce complexity and narrow down potential errors, please update to the latest release 3.5 (if not already done), rebuild the sick_scan_xd library and run [sick_scan_xd_api_test.py](https://github.com/SICKAG/sick_scan_xd/blob/develop/test/python/sick_scan_xd_api/sick_scan_xd_api_test.py) on your target:
```
export LD_LIBRARY_PATH=.:./build:$LD_LIBRARY_PATH
export PYTHONPATH=.:./python/api:$PYTHONPATH
python3 ./test/python/sick_scan_xd_api/sick_scan_xd_api_test.py ./launch/sick_lms_4xxx.launch hostname:=10.95.76.102 2>&1 | tee -a sick_scan_xd_api_test_lms_4xxx.log
```
The python script sick_scan_xd_api_test.py just initializes the lidar, registers callbacks for point cloud and log messages and prints a short text in the callbacks. Can you reproduce the error in this reduced test case? If so, please post a full logfile of the console messages. Is your target the NVIDIA Jetson Xavier as mentioned in #376?

spark-res, 2024-08-19:
Hi @rostest,

Appreciate the help so far.

Attached are the results from running the above. This is three runs, they all ended prematurely I think due to segmentation faults (not caught in logs) but visible in console.

[run_1_sick_scan_xd_api_test_lms_4xxx.log](https://github.com/user-attachments/files/16656656/run_1_sick_scan_xd_api_test_lms_4xxx.log)
[run_2_sick_scan_xd_api_test_lms_4xxx.log](https://github.com/user-attachments/files/16656657/run_2_sick_scan_xd_api_test_lms_4xxx.log)
[run_3_sick_scan_xd_api_test_lms_4xxx.log](https://github.com/user-attachments/files/16656658/run_3_sick_scan_xd_api_test_lms_4xxx.log)

The _SickScanApiSendSOPAS_ is causing a segmentation fault as per this issue https://github.com/SICKAG/sick_scan_xd/issues/376 - Could the TcpRecvThread issue be related? I'll run through the gdb steps and see where that leads too.

I commented out the _SickScanApiSendSOPAS_ command and it runs ok. I've run some longer tests with no _TcpRecvThread_ issue, but need to do more testing to rule it out as it's quite intermittent at times.

rostest, 2024-08-19:
Thanks for your reply. The logfiles do not show abnormalities. This issue can indeed be related to #376.

spark-res, 2024-08-22:
Hi @rostest,

Looks like were still seeing the _TcpRecvThread_ exiting early issue. Although it is looking like an interaction with the OS or other parts of our code. We don't see it in the _sick_scan_xd_api_test_ or even our own simplified examples.

It seems to reliably occur when we call a specific function in our overall application (a function that's working with a copy of the point cloud data). This function in theory shouldn't interact with the sick_scan_api thread or data.

We added the diagnostic call-back and these are the messages we see:

[image]

Is there any way to determine where the exit status code = 4 came from?

rostest, 2024-08-23:
Thanks for your reply. A diagnostic message with status code = 4 just means that sick_scan_xd is currently exiting. Just a thought: Did you make a deep copy of the scan data received in the python API callback? Note that the memory of the scan data buffer (type ctypes.POINTER) is released after the callback has been executed. Using a shallow copy instead of a deep copy may produce unexpected results. See https://github.com/SICKAG/sick_scan_xd/blob/develop/doc/sick_scan_api/sick_scan_api.md#usage-example for the memory layout and deep copies of the point cloud.

spark-res, 2024-08-26:
Thanks @rostest,

Confirming we are using a deep copy of the scan buffer. I've also ran the problematic function using hard-coded data, not from the lidar thread and the issue is still present.

It does seem like an interaction in the background (OS/HW/?) is causing the tcp thread to exit. I'll continue to do more troubleshooting in the background. Otherwise we may have to temporarily disable the problematic function as a workaround for the time being.

Babs13400, 2024-10-08:
Hello 

We encountered the same issue. And we noticed that is wasn't pretty fixed when the topic was closed 2 weeks ago.
Please can you reopen it in order to discuss about it.

Thank you very much

rostest, 2024-10-08:
@Babs13400 Thanks for your feedback. The issue is reopened for further discussion.

If you encounter issues with the sick_scan_xd driver, please provide more details, especially:
* What lidar and OS are you using?
* Does sick_scan_xd work as expected when using [sick_scan_xd_api_test.py](https://github.com/SICKAG/sick_scan_xd/blob/develop/test/python/sick_scan_xd_api/sick_scan_xd_api_test.py)?
   * If yes: Please describe the error in detail and provide the necessary information and sample code to reproduce the error.
   * If not: Please provide a complete logfile from start until the error occures.

rostest, 2024-10-09:
Thank you for this information. An unexpected termination of point cloud messages after a few hours can have several reasons.

Does the error also occur when running the small test program [sick_scan_xd_api_test.cpp](https://github.com/SICKAG/sick_scan_xd/blob/develop/test/src/sick_scan_xd_api/sick_scan_xd_api_test.cpp)? Does sick_scan_xd generate any error messages or warnings during the run, or can you provide a complete logfile from start to error?

Is the LMS511 still sending scan data telegrams after the error? Please use Wireshark to monitor the network traffic between the lidar and sick_scan_xd:
* Download and install Wireshark (installer or portable version) from https://www.wireshark.org/download.html
* Start Wireshark and select the interface for TCP communication with the lidar.
* Observe the network traffic - is the lidar still sending TCP packets to your Windows PC?

If possible, please send a screenshot.

Babs13400, 2024-10-28:
Hello
Thanks for your help. We successfully fixed our issue. It was related to the network switch used in our Sytem. 
Actually their was several devices that share the same switch and our LMS511used it as well.
And when we plug the scanner directly in the PC network board, we no longer observe PointCloudMsg paquet error.
Thank you for your Help.

Additional question before closing the topic :
I read some topics in the project telling that the API doesn't support multiple scanners using without ROS.
So my question is, how SOPAS achieve this ?
Actually when we use SOPAS, we can retrieve and display one or several scanners data at the same time.

Thank you very much

rostest, 2024-10-29:
Thanks for the information. To run multiple lidars simultaneously, we recommend using ROS or running sick_scan_xd in multiple and separate processes, such that each process serves one sensor. The reason for this is that the sick_scan_xd library is not fully thread-safe. See [Start multiple nodes](https://github.com/SICKAG/sick_scan_xd?tab=readme-ov-file#start-multiple-nodes) for more details.

Babs13400, 2024-10-29:
Hello
Thank you for your reply. So my question is how SOPAS ET achieves running
several Scanners at the same time.
Thank you

rostest, 2024-10-29:
SOPAS ET and sick_scan_xd are independent software programs with different approaches and implementations. Please contact SICK support for questions about SOPAS ET or feature requests for sick_scan_xd.

Babs13400, 2024-10-29:
Hello. Thanks for you reply. You can consider the topic completed for my
part.
Cheers

closed 2024-10-29 by rostest