SICKAG/sick_scan_xd #362: Changing the build/install directory

spark-res, 2024-07-17:
Hello,

We are working on a system that wipes the /usr/ folder after each reboot.

We would like to install the sick libraries in an alternate location, so they're not wiped on boot.

We're using Generic Linux without ROS. Is there any instructions to install to an alternate location?

https://github.com/SICKAG/sick_scan_xd/blob/develop/INSTALL-GENERIC.md#build-on-linux-generic-without-ros

Thanks,
Steven

rostest, 2024-07-17:
Thanks for your feedback. To use the sick_scan_xd library without ROS or installation, just copy the `libsick_scan_xd_shared_lib.so` file from the build directory to a persistent folder that is included in LD_LIBRARY_PATH. This way the library can be dynamically loaded at runtime by [sick_scan_xd_api_wrapper.c](https://github.com/SICKAG/sick_scan_xd/blob/develop/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c).

spark-res, 2024-07-23:
Thanks @rostest.

Confirming that copying it to a persistent directory and updating LD_LIBRARY_PATH is an acceptable and working solution.

closed 2024-07-23 by spark-res