SICKAG/sick_scan_xd #363: Error Spam in sick_generic_caller when using with MULS1AA-112211 multiScan136 model

Epik-Robotik, 2024-07-17:
Whenever we run this node with the 3d lidar the log is spammed with errors and warning continously that make it impossible to debug.
The spammed errors are:
`[INFO][2024-07-17 12:39:34][async_movaicore.py][_rosout_callback][name:/tp_navigation__tp_drivers__multiscan_udp__sick_scan|function:MsgPackExporter::RunCb|service:spawner][97]: MsgPack/Compact-Exporter:   0 udp packages still in input fifo, 0 messages still in output fifo, current segment index: 4
[INFO][2024-07-17 12:39:34][async_movaicore.py][_rosout_callback][name:/tp_navigation__tp_drivers__multiscan_udp__sick_scan|function:MsgPackExporter::RunCb|service:spawner][97]: MsgPack/Compact-Exporter: max. 21 udp messages buffered, max 21 export messages buffered.
[INFO][2024-07-17 12:39:34][async_movaicore.py][_rosout_callback][name:/tp_navigation__tp_drivers__multiscan_udp__sick_scan|function:MsgPackExporter::RunCb|service:spawner][97]: MsgPack/Compact-Exporter: 1447674 messages exported at 239.899 Hz, mean time: 1.754 milliseconds/messages, stddev time: 4.603, max time: 256.952 milliseconds between udp receive and messages export, histogram=[587632,668079,32804,24730,56138,49118,9873,4600,2773,1771,10156]
[WARNING][2024-07-17 12:39:34][async_movaicore.py][_rosout_callback][name:/tp_navigation__tp_drivers__multiscan_udp__sick_scan|function:MsgPackValidator::validateNoMissingScandata|service:spawner][99]: ## WARNING MsgPackValidator::validateNoMissingScandata() failed: no scan data found in echo 0:
[ERROR][2024-07-17 12:39:34][async_movaicore.py][_rosout_callback][name:/tp_navigation__tp_drivers__multiscan_udp__sick_scan|function:MsgPackConverter::Run|service:spawner][101]: ## ERROR MsgPackConverter::Run(): msgpack validation failed (scandata missing)
[ WARN] [1721219974.526825750]: ## WARNING MsgPackValidator::validateNoMissingScandata() failed: no scan data found in echo 0:
[ERROR] [1721219974.526911698]: ## ERROR MsgPackConverter::Run(): msgpack validation failed (scandata missing)
`

we tried changing some parameters but it doesnt seem to work.

rostest, 2024-07-17:
Thanks for your feedback. You can configure the verbosity of a ros node by setting a loglevel in the launchfile, e.g.:
`<env name="ROSCONSOLE_CONFIG_FILE" value="/tmp/rosconsole_loglevel_warn.conf" />`

This will configure the loglevel via file `/tmp/rosconsole_loglevel_warn.conf`. The loglevel is INFO by default. To set loglevel WARN (i.e. to display warnings and errors only), the file  `/tmp/rosconsole_loglevel_warn.conf` contains the line `log4j.logger.ros=WARN`. See https://wiki.ros.org/rosconsole#Configuration for further details.

Your messages show warnings `## WARNING MsgPackValidator::validateNoMissingScandata() failed`. This indicates that a MsgPackValidator is activated, which is a test option for the msgpack format only. Please use the default format `compact` for transmission of scan data (parameter scandataformat = 2 in the launchfile) and deactivate the MsgPackValidator (msgpack_validator_enabled = False). We recommend to use the most recent sick_scan_xd release 3.4. with default parameters.

Epik-Robotik, 2024-07-17:
The release 3.4. doesnt seem to be published on ubuntu server, could you please check.

rostest, 2024-07-18:
Thanks for your reply. apt provides sick_scan_xd release 3.4 for ROS2 humble and 3.2 for ROS1 noetic. Please feel free to use prebuilt release 3.2 for noetic or build 3.4 from sources as described in https://github.com/SICKAG/sick_scan_xd/blob/develop/INSTALL-ROS1.md.

The message `ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred` is not printed by sick_scan_xd.

closed 2024-08-15 by rostest