SICKAG/sick_scan_xd #366: Can't communicate with Sick Tim781

akashjinandra, 2024-07-19:
Hello, I'm running Ubuntu 20.04, with ROS 1 Noetic. I'm trying to talk to a Sick TIM781 sensor. I was able to ping the device, and then use SOPAS to change the communication to binary. But I still get this error

[ INFO] [1721348868.739022377]: sick_scan_xd: Tcp::open: connecting to 192.168.100.121:2112 ...
[ INFO] [1721348868.739646951]: sick_scan_xd Tcp::open: connected to 192.168.100.121:2112
[ INFO] [1721348868.739765460]: SickThread TcpRecvThread started.
[ INFO] [1721348868.740680908]: Parameter setting for <active_echo: 0>
[ INFO] [1721348868.742029990]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[ INFO] [1721348868.743799576]: Receiving: <STX>sRA SCdevicestate \x01<ETX>
[ INFO] [1721348868.743848127]: checkColaDialect: lidar response in configured Cola-dialect Cola-B
[ INFO] [1721348868.944213132]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ INFO] [1721348868.946663775]: Receiving: <STX>sAN SetAccessMode \x00<ETX>
[ INFO] [1721348869.147025367]: Sending  : <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>
[ INFO] [1721348869.148836878]: Receiving: <STX>sFA\x00\x0a<ETX>
[ WARN] [1721348869.148933552]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>, received answer: "sFA\x00\x0a", expected patterns: "sWA EIHstCola","sAN EIHstCola"
[ WARN] [1721348874.149309369]: Timeout during waiting for new datagram
[ WARN] [1721348874.149446382]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>, received answer: "", expected patterns: "sWA EIHstCola","sAN EIHstCola"
[ERROR] [1721348874.149554899]: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>, giving up after 1 unexpected answers.
[ INFO] [1721348874.149617679]: Failed to init scanner Error Code: 1
Waiting for timeout...
If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
2. Use the parameter "use_binary_protocol" to overwrite the default settings of the driver.
[ERROR] [1721348874.149687649]: ## ERROR in mainGenericLaser: init failed, retrying...
[ INFO] [1721348874.149724095]: Start initialising scanner [Ip: 192.168.100.121] [Port:2112]
[ WARN] [1721348874.149760133]: Disconnecting TCP-Connection.
[ERROR] [1721348874.150024228]: Tcp::readInputData: Read 0 bytes, connection is lost!
[ INFO] [1721348874.150129095]: SickThread TcpRecvThread finished (flags: threadShouldRun=0, endThread=0).
SickScanCommon closed.

rostest, 2024-07-22:
Thanks for your feedback and logfile. The driver establishes a TCP connection to the lidar and starts the initialisation using typical SOPAS commands. The SOPAS Command "sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44" should be answered by "sAN SetAccessMode 1", but is instaead acknowledged by "sAN SetAccessMode 0". As a consequence, the following  SOPAS command "sWN EIHstCola 1" for binary cola communication fails with error code 0x0A (Sopas_Error_VARIABLE_WRITE_ACCESSDENIED).

Most probably the password for the "authorized client" has been changed (default hash value is f4724744). Please change the password for "authorized client" back to the default, or reset your device to factory defaults and try again. 

[Tim781 logfiles](https://github.com/user-attachments/files/16333007/tim781.zip) with latest sick_scan_xd releases 3.4.0 and 3.5.0 and TiM781 firmware version V5.0 are attached for information.

akashjinandra, 2024-07-22:
Hello, thank you so much for your help. We have changed the passwords for our company for safety reasons. Do you know how to access the hash value for the password?

rostest, 2024-07-23:
@akashjinandra A direct calculation of the hash values is currently not available. We are clarifying further possibilities together with SICK.

akashjinandra, 2024-07-23:
Hello I was able to find away to get the calculated hash value from within the SOPAS ET software. You must be logged in as service level and you can calculate the the hash. However it seems like it did not seem to use the new value. I double checked that when I put the default password for level "authorized client" is "client". I was able to recreate the correct hash that is default in the launch file, however when I put the newly generated hash it doesn't seem to work.

rostest, 2024-07-26:
Thanks for new informations! Which version of SOPAS ET do you use? Can you post a screenshot? This topic may also be of interest to other users.

We will soon publish an update to support configurable hashvalues.

rostest, 2024-07-26:
@akashjinandra Please use the update in branch https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2407xx to configure hash values for the client authorization in the launchfile.

closed 2024-08-15 by rostest