SICKAG/sick_scan_xd #371: Some questions about the provided Python API

angxiaoming-666, 2024-07-25:
How can I control some operations of Sick in the provided Python API, such as setting its IP, output interval, etc., especially the start measure, stop, standby in the measurement.Looking forward to your reply, thank you

rostest, 2024-07-26:
Thank you for your feedback. Use SOPAS ET or SOPAS Air to configure settings such as the IP address. Lidar dependent scan settings like scan frequency or resolution can be configured via launchfile, see https://github.com/SICKAG/sick_scan_xd/blob/develop/USAGE.md for further information. Direct operational SOPAS commands at runtime are currently not supported by the API.

angxiaoming-666, 2024-07-26:
Thank you for your answer. I have two more questions to ask
1: I currently only see the Python/c++API, but I noticed some traces of Java programs in SOPAS ET. Is there any Java related API for Sick? Because I am currently using Java to call DLLs through JNA and writing code logic based on Python APIs‘
2: Actually, whether it's C++or Python APIs, they all use sockets at the bottom level. However, converting the frame information I obtain through sockets to coordinate information (x, y, z) is quite troublesome. Can you provide the relevant conversion logic?

rostest, 2024-07-26:
Thank you for your reply. Java is currently not directly supported. The Python/C++ examples may be helpful for loading the dll and calling the API functions via JNA.

Sockets are used for internal TCP communication with the lidar at a low level of the driver. They should not be used by the calling application. What frame information do you want to get from the socket?

closed 2024-09-25 by rostest