SICKAG/sick_scan_xd #375: Regarding the setting issue of the picoscan150 LiDAR's angle range.

wdsrsy, 2024-08-07:
Hello engineers, I have encountered an issue with the picoscan150. I wanted to change the angle range while using it, so I modified two options, `all_segments_min_deg` and `all_segments_max_deg`, in the `sick_picoscan.launch` configuration file. However, when I checked with the RViz2 tool, I found no effect; it still displays the full range of -138 to +138 degrees. The same is true when printing out the `/scan_fullframe` topic. I would like to ask, does the picoscan150 not support changing the angle range through configuration?

rostest, 2024-08-07:
Thanks for your feedback. The picoScan 150 is available in multiple variants. According to my understanding, all models have a fixed horizontal aperture of 276 degree. An angle range filter is not supported. Open the lidars ip address in a browser and use SOPAS Air to check and configure possible settings.

wdsrsy, 2024-08-08:
Thank you, I have also found that using the SOPAS tool can make changes. However, after I made the changes, I found another issue: when I re-ran the picoscan150 driver, I found that there were topics but no topic data came out.

rostest, 2024-08-08:
Thank you for your reply. The topic "/sick_multiscan/scan" is obsolete and will be removed in the next release. It can simply be ignored.

wdsrsy, 2024-08-08:
Oh, right, I changed the launch configuration parameters to the range set under SOPAS, and now I can get the topic data.

closed 2024-08-15 by rostest