SICKAG/sick_scan_xd #377: No topic published after launch with no error found for picoscan150

Phymin, 2024-08-08:
Hi,

We're using ros2 humble and picoscan150 for SLAM navigation, currently we met an issue: first, the sick_driver_scan node is crashed (we don't find the reason), then we launch the sick_scan_xd node manually again, though we don't see any error out, but there is also no data published, and with tcpdump, there is also no data coming from port 2115, but from SOPAS air, we can see the point cloud normally. We tried to reboot the IPC (lidar is not powered off), but it doesn't work, only after we power off and power on the robot, this issue was resloved. I have attached the launch logs below, please give us some suggestions of how to solve this issue. Thanks
[sick_launch failed.txt](https://github.com/user-attachments/files/16541852/sick_launch.failed.txt)

rostest, 2024-08-08:
Thank you for your feedback and log file. The logfile shows successful TCP communication and lidar initialisation, but no scan data.

Picoscan lidars send their scan data via UDP to the receiving PC, i.e. the sick_scan_xd driver. A common error is an incorrect configuration of the receiver's IP address. Please check the udp_receiver_ip parameter in the launch file. If tcpdump or wireshark does not observe any UDP packets from the lidar, the scan data is most likely being sent to the wrong ip address. Make sure that "192.168.3.10" is the correct ip address of your IPC running sick_scan_xd. Use the same receiver ip as in SOPAS Air. See https://github.com/SICKAG/sick_scan_xd/blob/develop/doc/sick_scan_segment_xd.md#configuration for more information.

The sick_scan_xd driver is self-monitoring and will exit with a timeout if no scan data is received. The timeout can be configured in the launch file with the parameter udp_timeout_ms (timeout for udp messages in milliseconds, default: 60000).

Phymin, 2024-08-09:
Hi,

Thanks for your feedback, but I don't think it's a configuration issue, if it is, this issue should happen all the time, but not occasionaly. We just don't understand why should we do power restart to resolve this issue, and we also encountered this issue in Tim240 lidar before. Only after power restart, we can receive the lidar point topic. I have also attached the launch file we used, you can check if there is something wrong in it. Thanks
[sick_picoscan150.launch.txt](https://github.com/user-attachments/files/16555555/sick_picoscan150.launch.txt)

rostest, 2024-08-09:
Thanks for your reply and further information. If it happens occasionally, it's certainly different.

If the problem persists after a warm start (i.e. after rebooting the IPC), but requires a cold start (i.e. it works again after powering down and up), it looks more like a hardware problem than a software issue. Maybe the network adapter requires a power cycle to restart and reconnect, but this is just a guess.

For further network diagnostics, please run the following steps on the IPC after the problem occurs:

* Check if the lidar can be reached with `ping 192.168.3.30`. If not, please check the network settings and the ethernet adapter for a system or hardware problem. If the lidar is not pingable other than through a power cycle, there is most likely a hardware related network problem.
* Start tcpdump or wireshark. Can you see any UDP or TCP traffic between the IPC and the lidar?
* Kill sick_scan_xd (`kill -9 sick_generic_caller`) and check again: Is the lidar pingable, and if so, is it sending UDP packets?
* Relaunch sick_scan_xd. Can a TCP connection be established? If so, are there any error messages from the driver? Is the lidar still pingable otherwise?

 Your picoScan launch file shows modified settings for the parameters `all_segments_min_deg` and `all_segments_max_deg` (-105 to +105 instead of -138.0 to +138.0 degrees). The picoScan lidars have a horizontal aperture of 276 deg. The parameters all_segments_min_deg and all_segments_max_deg in the launchfile should therefore be -138.0 and +138.0 degrees respectively. We recommend to using `<arg name="all_segments_min_deg" default="-138.0" />` and `<arg name="all_segments_max_deg" default="+138.0" />`.

Phymin, 2024-08-12:
Hi,

To your questions, when this issue happened, we can ping the lidar with 192.168.3.30, and with tcpdump, there is no data comming from UDP port 2115. 

The reason we use -105~+105 degrees is because using full range of lidar FOV will result in seeing the robot itself. 

We will provide more information if this issue happens again.

closed 2024-09-25 by rostest