SICKAG/sick_scan_xd #383: Read Error when getting datagram: 1

hwulfmeyer, 2024-08-19:
Hello,

I am getting the following errors when running the C++ minimum_sick_scan_api_client with our LRS4000 (Firmware version
1.6.0.25R)

```
[Error]: Read Error when getting datagram: 1
[Error]: Read Error when getting datagram: 1
[Error]: ## ERROR SickScanMonitor::checkState(): read timeout after 30.0008 sec, timeout (15 sec) exceeded.
```
full logs:
https://gist.github.com/hwulfmeyer/e5105643a1354a389d6d6e8b7bc9ee18

Note: I upped the timeout to 15s from the default 5s and upped the sleep in the C++ test to 60s. With the default timeout and sleep values the same error(s) occur.

Any help would be massively appreciated. Thank you.

rostest, 2024-08-20:
Thanks for your feedback. The logfile shows a successful lidar initialization, but no scan data is being received. Can you see the pointcloud with [SOPAS ET](https://www.sick.com/no/en/catalog/products/lidar-and-radar-sensors/lidar-sensors/lrs4000/c/g555594?category=g569793&tab=downloads)

In rare cases the start time required to fully change into operational mode might exceed 30 seconds. Please increase the timeouts to 120 seconds in the launchfile and run the test over a longer period:
```
        <param name="read_timeout_millisec_default" type="int" value="120000"/>
        <param name="read_timeout_millisec_startup" type="int" value="120000"/>
        <param name="read_timeout_millisec_kill_node" type="int" value="150000"/>
```

hwulfmeyer, 2024-08-22:
The problem was, that in the settings of the LiDAR the measurement data output was set to the wrong protocol. "Compact" instead of "LMDscandata".

closed 2024-08-22 by hwulfmeyer

rostest, 2024-08-23:
Thank you very much for this interesting and important information!