SICKAG/sick_scan_xd #385: can't run  sick_generic_caller.exe

Ofurosuki, 2024-09-09:
Hi, when I try to run sick_generic_caller.exe on PowerShell, the software stops because of debug assertion error. The error message is the following.
[image]
[image]
I compiled it following the instructions for 'build for Windows'(https://github.com/SICKAG/sick_scan_xd/blob/develop/INSTALL-GENERIC.md#build-on-windows). I ran these commands on Developer Command Prompt for VS 2019.
.
What do you think I should do?

environment:
windows10 pro
Visual Studio 2019

Ofurosuki, 2024-09-09:
Thank you for your quick response and please excuse my lack of explanation. I use picoscan150 pro. So, my command is
```
sick_generic_caller.exe absolute\adress\to\sick_picoscan.launch hostname:=192.168.0.1 udp_receiver_ip:=192.168.0.2
```
I confirmed that both the IP addresses are correct.

rostest, 2024-09-10:
Thank you for the information. We fixed this issue in the branch [feature/issue_update_2407xx](https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2407xx). Please checkout the latest update (`git clone -b feature/issue_update_2407xx https://github.com/SICKAG/sick_scan_xd.git`) and rebuild sick_scan_xd.

Ofurosuki, 2024-09-10:
It worked. I appreciate your help so much.

closed 2024-09-10 by Ofurosuki

rostest, 2024-09-10:
@Ofurosuki Thank you for reporting!