SICKAG/sick_scan_xd #386: Build Failure on Raspberry Pi OS Bookworm, RasPi 4, ROS 2 Jazzy

pgadamczyk, 2024-09-10:
I have managed to build ROS 2 Jazzy on a Raspberry Pi 4 with Raspberry Pi OS Bookworm. I am attempting to follow the build instructions for Raspberry Pi, but the build hangs and crashes at about 75%. Is this a known issue? 

I issued this command: 
`colcon build --packages-select sick_scan_xd --cmake-args " -DROS_VERSION=2" " -DLDMRS=0" " -DRASPBERRY=1" " -DSCANSEGMENT_XD=0" --event-handlers console_direct+`

And after lots of success getting to around 75%, it generated this error (from the events.log file): 

```
...
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.                                            
gmake[2]: *** [CMakeFiles/sick_scan_xd_lib.dir/build.make:286: CMakeFiles/sick_scan_xd_lib.dir/driver/src/sick_scan_common.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:630: CMakeFiles/sick_scan_xd_lib.dir/all] Error 2
...
```

Help getting it unstuck would be much appreciated! Happy to share more complete logs.

rostest, 2024-09-10:
Thanks for your feedback. It looks like gcc is aborting. This is not a known problem, perhaps Raspberry support or the [Raspberry Pi OS forum](https://forums.raspberrypi.com/viewforum.php?f=66) can help. Repeating the colcon command after the crash should continue the build. Is there a specific file that is causing gcc to exit?

closed 2024-09-25 by rostest

pgadamczyk, 2024-09-30:
Thanks for following up. I had to turn away for a while, but turned back
today.
The build was crashing at random times, not some specific package, which
led me to suspect a memory problem (I'm using a RasPi 4, 4GB model).
I was able to succeed today by calling a single-core colcon build (
MAKEFLAGS=-j1 ):

MAKEFLAGS=-j1 colcon build --packages-select sick_scan_xd --cmake-args "
-DROS_VERSION=2" " -DLDMRS=0" " -DRASPBERRY=1" --event-handlers
console_direct+

I appreciate the attention.
-Peter